OpenC3/cosmos #1793: Can't open long edited suite, AXIOS error

tsjordan-eng, 2024-12-20:
**Describe the bug**
We have a long test suite (~600 lines) which runs over hours that we will edit portions of and re-run. Sometimes, however, when we try to open it (we see the asterisk denoting it has been edited from the plugin version) we get an axios error and it won't load. We do have local mode enabled so  when we hit this error, we move the file out of targets modified and then we can open the original plugin file.

[image]

This doesn't seem to happen on shorter files.

the docker container runs with the same user as owns the targets modified directories

**To Reproduce**
Steps to reproduce the behavior:

1. In script runner, edit a long file (\~600 lines)
2. run it for awhile (\~20 min to hour). Maybe disconnect network and reconnect
3. close tab, open new script runner tab
4. Try to open the file
5. See error above

**Expected behavior**
I should be able to open, edit, and run the file with no errors

**Environment (please complete the following information):**

- OS: RHEL9
- OpenC3 COSMOS Version 5.20.0
- Browser: Edge

jmthomas, 2024-12-20:
When this happens I'd like to know what is the original file name (and path), what you have in your local mode `plugins` folder, and what is in Bucket Explorer in `targets_modified`. In the meantime we'll see if we can reproduce.

ryanmelt, 2024-12-21:
A 500 error means there was an error in the backend.   Can you see if there are any stack traces in the scriptrunner container logs or cmd-tlm-api container logs when this happens.  "docker logs <container>"

tsjordan-eng, 2025-01-03:
Here is the relevant JSON log with the path:
<details>
```
{
  "time": 1735919664459239121,
  "@timestamp": "2025-01-03T15:54:24.459239Z",
  "level": "INFO",
  "microservice_name": "SCRIPT__RUNNER__API",
  "user": "anonymous",
  "container_name": "2ad7580d52d7",
  "message": "Completed #body",
  "type": "log",
  "thread_name": "puma srv tp 002",
  "duration": 2.2364606857299805,
  "request_id": "fb3a0538-d3a1-4e7f-9839-ef323bb28634",
  "path": "/script-api/scripts/TARGET_NAME/procedures/conops_test_long.rb*",
  "status": 500,
  "controller": "ScriptsController",
  "action": "body",
  "format": "HTML",
  "method": "GET",
  "allocations": 375,
  "view_runtime": 0.0,
  "exception_message": "\"\\xC2\" from ASCII-8BIT to UTF-8",
  "exception_class": "Encoding::UndefinedConversionError",
  "exception_backtrace": [
    "/usr/lib/ruby/3.2.0/delegate.rb:349:in `write'",
    "/usr/lib/ruby/3.2.0/delegate.rb:349:in `block in delegating_block'",
    "/src/app/models/script.rb:72:in `process_suite'",
    "/src/app/controllers/scripts_controller.rb:75:in `body'",
    "/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.4.2/lib/action_controller/metal/basic_implicit_render.rb:6:in `send_action'",
    "/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.4.2/lib/abstract_controller/base.rb:224:in `process_action'",
    "/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.4.2/lib/action_controller/metal/rendering.rb:165:in `process_action'",
    "/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.4.2/lib/abstract_controller/callbacks.rb:259:in `block in process_action'",
    "/usr/lib/ruby/gems/3.2.0/gems/activesupport-7.1.4.2/lib/active_support/callbacks.rb:101:in `run_callbacks'",
    "/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.4.2/lib/abstract_controller/callbacks.rb:258:in `process_action'",
    "/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.4.2/lib/action_controller/metal/rescue.rb:25:in `process_action'",
    "/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.4.2/lib/action_controller/metal/instrumentation.rb:74:in `block in process_action'",
    "/usr/lib/ruby/gems/3.2.0/gems/activesupport-7.1.4.2/lib/active_support/notifications.rb:206:in `block in instrument'",
    "/usr/lib/ruby/gems/3.2.0/gems/activesupport-7.1.4.2/lib/active_support/notifications/instrumenter.rb:58:in `instrument'",
    "/usr/lib/ruby/gems/3.2.0/gems/activesupport-7.1.4.2/lib/active_support/notifications.rb:206:in `instrument'",
    "/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.4.2/lib/action_controller/metal/instrumentation.rb:73:in`process_action'",
    "/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.4.2/lib/action_controller/metal/params_wrapper.rb:261:in `process_action'",
    "/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.4.2/lib/abstract_controller/base.rb:160:in `process'",
    "/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.4.2/lib/action_controller/metal.rb:227:in `dispatch'",
    "/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.4.2/lib/action_controller/metal.rb:309:in `dispatch'",
    "/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.4.2/lib/action_dispatch/routing/route_set.rb:49:in `dispatch'",
    "/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.4.2/lib/action_dispatch/routing/route_set.rb:32:in `serve'",
    "/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.4.2/lib/action_dispatch/journey/router.rb:51:in `block in serve'",
    "/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.4.2/lib/action_dispatch/journey/router.rb:131:in `block in find_routes'",
    "/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.4.2/lib/action_dispatch/journey/router.rb:124:in `each'",
    "/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.4.2/lib/action_dispatch/journey/router.rb:124:in `find_routes'",
    "/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.4.2/lib/action_dispatch/journey/router.rb:32:in `serve'",
    "/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.4.2/lib/action_dispatch/routing/route_set.rb:882:in `call'",
    "/usr/lib/ruby/gems/3.2.0/gems/rack-3.1.8/lib/rack/etag.rb:29:in `call'",
    "/usr/lib/ruby/gems/3.2.0/gems/rack-3.1.8/lib/rack/conditional_get.rb:31:in `call'",
    "/usr/lib/ruby/gems/3.2.0/gems/rack-3.1.8/lib/rack/head.rb:15:in `call'",
    "/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.4.2/lib/action_dispatch/middleware/callbacks.rb:29:in `block in call'",
    "/usr/lib/ruby/gems/3.2.0/gems/activesupport-7.1.4.2/lib/active_support/callbacks.rb:101:in `run_callbacks'",
    "/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.4.2/lib/action_dispatch/middleware/callbacks.rb:28:in `call'",
    "/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.4.2/lib/action_dispatch/middleware/debug_exceptions.rb:29:in `call'",
    "/usr/lib/ruby/gems/3.2.0/gems/actionpack-7.1.4.2/lib/action_dispatch/middleware/show_exceptions.rb:31:in `call'",
    "/usr/lib/ruby/gems/3.2.0/gems/rails_semantic_logger-4.17.0/lib/rails_semantic_logger/rack/logger.rb:45:in `call_app'",
    "/usr/lib/ruby/gems/3.2.0/gems/rails_semantic_logger-4.17.0/lib/rails_semantic_logger/rack/logger.rb:26:in `block in call'",
    "/usr/lib/ruby/gems/3.2.0/gems/semantic_logger-4.16.1/lib/semantic_logger/base.rb:190:in `block in tagged'",
    "/usr/lib/ruby/gems/3.2.0/gems/semantic_logger-4.16.1/lib/semantic_logger/semantic_logger.rb:395:in `named_tagged'",
    "/usr/lib/ruby/gems/3.2.0/gems/semantic_logger-4.16.1/lib/semantic_logger/base.rb:197:in `tagged'",
    "/usr/lib/ruby/gems/3.2.0/gems/rails_semantic_logger-4.17.0/lib/rails_semantic_logger/rack/logger.rb:26:in `call'",
[... cut ...]
  "@timestamp": "2025-01-03T15:54:24.459868Z",
  "level": "FATAL",
  "microservice_name": "SCRIPT__RUNNER__API",
  "user": "anonymous",
  "container_name": "2ad7580d52d7",
  "message": "Exception was raised - Encoding::UndefinedConversionError:\"\\xC2\" from ASCII-8BIT to UTF-8",
  "type": "log",
  "thread_name": "puma srv tp 002",
  "request_id": "fb3a0538-d3a1-4e7f-9839-ef323bb28634"
}
```
</details>

I am only ever editing the file from the COSMOS Script Runner Editor, never externally so I don't know why there's some conversion error.

tsjordan-eng, 2025-01-03:
Gem path: `targets/TARGET/procedures/conops_test_long.rb`
Bucket Explorer path: `/targets_modified/TARGET_NAME/procedures/`, `conops_test_long.rb`
Local mode path: `plugins/DEFAULT/targets_modified/TARGET_NAME/procedures/conops_test_long.rb`
Script Runner JSON Log path: `/script-api/scripts/TARGET_NAME/procedures/conops_test_long.rb*`

The one downloaded from the bucket explorer is identical to the one in local mode.

PR #1853 (2025-01-24): Ensure target_file returns utf-8 files
closes #1793
closed 2025-01-24 by ryanmelt