OpenC3/cosmos #1794: Can an accessor operate within another?

tsjordan-eng, 2024-12-20:
The accessors are a very powerful tool for decommutating packets. However, what if I have a binary telemeter that has a string item which I know to be json? Can I break out that string with an accessor?

Some alternatives may be:
- using derived items with ruby that parses the json
- pushing that packet item out a router and having another interface/target access it

Any other ideas?

jmthomas, 2025-01-06:
We're doing something similar with the `http_accessor` which uses a internal `form_accessor` to access the internal items. Unfortunately we don't have a good generic solution to this problem so I'll leave this open as an enhancements. I think @ryanmelt has some ideas ...

ryanmelt, 2025-01-06:
So you can do this already using object_read_conversion.   However it will create one item that returns a hash.
I'm coming up with some methods to make the hash usable with some other tickets.

closed 2025-02-23 by ryanmelt