OpenC3/cosmos #1796: Height vs Padding discrepancies in Pixel values

mountTom, 2024-12-24:
**Describe the bug**
In the Telemetry viewer screen, the setting PADDING vs HEIGHT have different VALUES for pixels 

**To Reproduce**
create the following screen.

`
SCREEN AUTO AUTO 1.0
GLOBAL_SETTING LABEL BACKCOLOR BLUE

MATRIXBYCOLUMNS 3

LABEL "TEST"
SETTING PADDING 100
LABEL "TEST 2"
SETTING HEIGHT 100
LABEL "TEST 3"
SETTING HEIGHT 225
`

**Expected behavior**
You would expect the label of TEST , and TEST2 to be the same Height. The documentation describes that both default values are size in pixels.

THIS is not the case. In this example, The "pixel size" of the HEIGHT setting is 2.25 times smaller than the PADDING setting. This SCALER changes with overall  size of the window.

**Screenshots**
[image]

**Environment (please complete the following information):**

- OS: [Ubuntu 20.04]
- OpenC3 COSMOS Version [6.0.1]
- Browser: [Firefox]

jmthomas, 2024-12-24:
PADDING of 100 means padding all around the text. So TEST is approximately twice the height of TEST 2 which is just specifying the HEIGHT.

mountTom, 2024-12-26:
Any reason it is not exactly twice the hight?

mountTom, 2024-12-26:
It is the size of the text! Thank you for the help!

closed 2024-12-26 by mountTom