OpenC3/cosmos #1801: Transferring several megabyte block of data using template accessor

mkalin2, 2024-12-30:
I am communicating with an oscilloscope using a SCPI command/response protocol. All off the commands/telemetry packets to/from the scope are very simple and work well in the CmdResponse protocol.

However there is a waveform transfer command which will send the raw bytes from the scope capture like this:

`CH1:WF ALL,<WAVEFORM_DATA>`

Where the WAVEFORM_DATA could be several megabytes. In the tlm file WAVEFORM_DATA is sized to be 2GB (although I'm not sure if the template protocol even uses those bit sizes).

When I run my waveform transfer command, I see the expected number of bytes in the Cmd/Tlm server and when I can also confirm that all the expected bytes made it into COSMOS when I scroll down to the bottom of the packet in the raw packet viewer. However, when I try to use the data extractor or the `tlm()` command in a script to parse the packet data. The data is truncated down to ~3.7MB when it should be ~5MB in this case. 

Can COSMOS handle packets this big? Is this a limitation of the template protocol? There is a length field I in the response packet that I could use, but it is only in this packet and not the rest of my oscilloscope packets. Is there a way to use the length protocol for only this packet?

mkalin2, 2024-12-30:
Cmd/Tlm server view:
[image]

Checking the length of the data in a script:
[image]

mkalin2, 2024-12-30:
I also just tried this with a much smaller waveform (25 kB) and the data is still truncated (down to 19 kB). Seems to be a rather consistent 75% truncation?

ryanmelt, 2024-12-30:
Are there null bytes in the data 0x00?
Are you declaring the field as a STRING or a BLOCK?
Also don't hard code a 2GB field, that will waste tons RAM.   Please make it variable sized (bit size of 0)

mkalin2, 2024-12-30:
I didn't know 0 could be used for variable sized items, thanks.

Yes there are null bytes in the data and it is declared as a BLOCK. The null bytes are making it into COSMOS, at least at the beginning.

ryanmelt, 2024-12-30:
Will see if we can recreate.  Seems like you are running into a 5MB hard coded size limit somewhere in our HTTP stack.

mkalin2, 2025-01-02:
I have investigated this some more. The point where the data is truncated is the index of the first newline character (byte = 0x0A) in the data block. If I capture a waveform which doesn't have any bytes equal to 0x0A there are no issues. 

I purposely am not using the terminated protocol to ensure that these bytes don't cause my packet to get cut short when receiving. Does it make sense that the data extractor or `tlm` command would still terminate a block upon seeing a newline character?

ryanmelt, 2025-01-02:
Can you provide the packet definition and Interface/Protocol configuration?

mkalin2, 2025-01-02:
Here is my plugin.txt:
```
TARGET OSCILLOSCOPE OSCILLOSCOPE

INTERFACE OSCILLOSCOPE_INTERFACE tcpip_client_interface.rb 192.168.44.11 1861 1861 10.0 nil
  MAP_TARGET OSCILLOSCOPE
  PROTOCOL READ_WRITE VicpProtocol
  PROTOCOL READ_WRITE CmdResponseProtocol 10
```
VicpProtocol is a very simple custom protocol based on terminated protocol that removes the final byte of the returned data (which is always a newline character) since I cant use terminated protocol as explained above. Literally `packet_data = @data[0..(length-2)]`. I have checked with Logger that the length of packet data before it gets returned is 5MB, so I don't think its an issue in this protocol, but let me know if you think there are other things I should check.

Packet definition:
```
TELEMETRY OSCILLOSCOPE WAVEFORM_RESPONSE BIG_ENDIAN
  ACCESSOR TemplateAccessor
  TEMPLATE "<CHANNEL_NUM>:WF ALL,<WAVEFORM_DATA>"
  APPEND_ITEM CHANNEL_NUM 16 STRING
  APPEND_ITEM WAVEFORM_DATA 0 BLOCK
```
Here is how the packet looks in the Cmd/Tlm server raw viewer:
```
Address   Data                                             Ascii
---------------------------------------------------------------------------
00000000: 4d 38 3a 57 46 20 41 4c 4c 2c 23 39 30 30 35 30  M8:WF ALL,#90050
00000010: 30 30 33 34 36 57 41 56 45 44 45 53 43 00 00 00  00346WAVEDESC   
00000020: 00 00 00 00 00 4c 45 43 52 4f 59 5f 32 5f 33 00       LECROY_2_3 
00000030: 00 00 00 00 00 01 00 01 00 5a 01 00 00 00 00 00           Z      
00000040: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00                  
00000050: 00 40 4b 4c 00 00 00 00 00 00 00 00 00 00 00 00   @KL            
00000060: 00 4c 45 43 52 4f 59 57 52 38 31 30 38 48 44 00   LECROYWR8108HD 
00000070: 00 53 f1 00 00 00 00 00 00 00 00 00 00 00 00 00   S              
00000080: 00 00 00 00 00 a0 25 26 00 a0 25 26 00 a0 25 26        %&  %&  %&
```
Here is where the first newline character is that breaks the data extraction:
```             
003a0df0: 0d 08 0d 08 0d 08 0d 08 0d 08 0d 08 0d 08 0d 08                  
003a0e00: 0d 08 0d 08 0d 08 0d 08 0d 08 0d 08 0d 08 0d 08                  
003a0e10: 0d 08 0d 08 0d 08 0d 08 0d 08 0d 08 0d 08 0d 08                  
003a0e20: 0d 08 0d 08 0d 08 0d 08 0d 08 0d f4 0a f4 0a f4   <--- this line                  
003a0e30: 0a f4 0a f4 0a f4 0a f4 0a f4 0a f4 0a f4 0a f4                  
003a0e40: 0a f4 0a f4 0a f4 0a f4 0a f4 0a f4 0a f4 0a f4                  
```
Here is the end of the packet. We see all the data after all that newline, as expected (at least in the raw viewer):
```
004c4c40: a7 5e a7 5e a7 5e a7 5e a7 5e a7 5e a7 5e a7 5e   ^ ^ ^ ^ ^ ^ ^ ^
004c4c50: a7 5e a7 5e a7 5e a7 5e a7 5e a7 5e a7 5e a7 5e   ^ ^ ^ ^ ^ ^ ^ ^
004c4c60: a7 5e a7 5e a7 5e a7 5e a7 5e a7 5e a7 5e a7 5e   ^ ^ ^ ^ ^ ^ ^ ^
004c4c70: a7 5e a7 5e a7 5e a7 5e a7 5e a7 5e a7 5e a7 5e   ^ ^ ^ ^ ^ ^ ^ ^
004c4c80: a7 5e a7 5e a7 5e a7 5e a7 5e a7 5e a7 5e a7 5e   ^ ^ ^ ^ ^ ^ ^ ^
004c4c90: a7 5e a7 5e a7 5e a7 5e a7 5e a7 5e a7 5e a7 5e   ^ ^ ^ ^ ^ ^ ^ ^
004c4ca0: a7 5e a7 5e a7 5e a7 5e a7 5e a7 5e a7 5e a7      ^ ^ ^ ^ ^ ^ ^ 
```
This final address `004c4ca0` corresponds to the expected ~5MB. I assume the fact that I can see this in the raw viewer means that the protocol is returning all of the data correctly, but I'm not sure how/when the raw packet viewer gets populated.

ryanmelt, 2025-01-02:
The newline is breaking the TemplateAccessor.  It is expecting everything to be on one line.

ryanmelt, 2025-01-02:
And plain text... This is the first use I've seen with binary data

ryanmelt, 2025-01-02:
For this packet I would recommend using the default BinaryAccessor.
If your channel number is always 2 bytes, then you can declare that as a two byte string.

mkalin2, 2025-01-03:
Ok yep that works.

Sorry I didn't realize you could have variable sized items with the BinaryAccessor.

closed 2025-01-30 by mkalin2