OpenC3/cosmos #1808: Setting the value of a TEXTBOX

mountTom, 2025-01-03:
I am trying to set the value of a textbox via a button. 
In previous versions of cosmos, there was a 
`screen.getNamedWidget("testing_box").setCurrentText("test")`
method but even 
`screen.getNamedWidget("testing_box").text() = "test"`

seemed to work.
I have also tried `.value` to no avail.

Is there a new method that I cannot find? 
Is there another place I can look to find potential methods? It seems like none of them are in the main repo?

jmthomas, 2025-01-07:
TEXTBOX widgets aren't designed to be used as user input. You should use a TEXTFIELD widget instead. Here are all the interactive widgets: https://docs.openc3.com/docs/configuration/telemetry-screens#interactive-widgets

mountTom, 2025-01-07:
Even If I change the value to a TEXTFIELD, I still cannot find a way to SET the text inside of it. I can GET with .text, but I am unable to SET.

closed 2025-01-08 by jmthomas

mountTom, 2025-01-08:
I was not using the var define! Thank you!