OpenC3/cosmos #1813: All time pickers can be set to UTC time

321github123, 2025-01-07:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Is your feature request related to a problem? Please describe.**
Our users use UTC instead of local time

**Describe the solution you'd like**
A global flag sets time pickers to use UTC instead of local time s e.g. Data Viewer, Data Extractor, Command History etc

**Additional context**
Add any other context or screenshots about the feature request here.

ryanmelt, 2025-01-07:
This already exists in Admin -> Settings.   
Is it not working somewhere?

321github123, 2025-01-08:
Does the Admin -> Settings change how the times are interpreted?  For example, will the times pickers in Data Extractor be now interpreted as UTC time vs local time if UTC is chosen in the Admin setting?

[image]

The help text in the Admin panel makes it seem that the Time Zone setting only affects the display of time.

[image]

ryanmelt, 2025-01-08:
It does only affect display on the frontend.  Our api and the backend is always UTC for everything.

321github123, 2025-01-08:
ok that makes sense -- can we add a UTC label to the time pickers then?

jmthomas, 2025-01-08:
Are you displaying the Astro Clock? The Astro Clock changes to UTC time which is present on all the tools.

jmthomas, 2025-01-08:
We have a lot of time pickers throughout our application that would have to manually add the UTC notification so this is unlikely to get implemented unless directly driven by customer requirements. I don't see anything in AstroUX that indicates adding 'UTC' to time pickers. I feel like the Astro Clock is a clear indicator for what the time format is.

321github123, 2025-01-08:
I think the confusion stemmed from seeing the local time in the AstroClock (since that is the default) and not knowing all time pickers are actually UTC.

Maybe a compromise is to default to UTC time zone?

321github123, 2025-01-13:
That's very helpful information.  Maybe we could add that information (setting the time zone sets the time pickers's time zone) here?

[image]