OpenC3/cosmos #1815: Shortening Long button code in JS

mountTom, 2025-01-08:
I currently have a button, that reads the values from multiple widgets, does some calculations, and sets the values of other widgets. I am currently doing this all in JS string format behind the button, which is rapidly increasing the size of my SCREEN file. 

I am aware of the runScript command, however in this case, since I require input from other widgets, I cannot offload the code to another file using ruby/python. 

Is there a way to reference JS code in the button script? 

For instance,
`BUTTON 'Run Script' "INST/procedures/testscript.js" `

So I can reference the widgets and clean up the screen?

Or is there better way to segment the screen?

ryanmelt, 2025-01-08:
You should be able to use partials like we do with packet definitions for things like headers.   
Create another file that starts with an underscore, put the button code in that, and then render it into the screen file.

mountTom, 2025-01-09:
Looking for some clarification here. From what I gather if I wanted to segment the following screen:
`
SCREEN AUTO AUTO 1.0
`
`
NAMED_WIDGET NEW TEXTFIELD 20 "TEST"
`
`
BUTTON 'CHANGE WIDGET'  		'var button = screen.getNamedWidget("NEW").value = "TEST Changed"' 
`

This has a textfield, and a button. where the button changes the text field. 

If I wanted to move the button code, I would (in the same directory), make a file named something like "_button.txt" . Then in the SCREEN render the file with ERB?

something like 

`
SCREEN AUTO AUTO 1.0
`
`
NAMED_WIDGET NEW TEXTFIELD 20 "TEST"
`
`
BUTTON 'CHANGE WIDGET'  		'<%= _button.txt%>'
`

with '_button.txt' =
`
'var button = screen.getNamedWidget("NEW").value = "TEST Changed"'
`

I have tried this with no success, where am I going wrong?

mountTom, 2025-01-09:
Got it, needed the line to be
`<%= render _button.txt%>`

Thank you for the help!

closed 2025-01-09 by mountTom