OpenC3/cosmos #1854: Telemetry viewer screens no longer editable (6.0.2)

mountTom, 2025-01-21:
**Describe the bug**
Can no longer see screen code, in the edit tab for Telemetry viewer

**To Reproduce**
1. Go to telemetry viewer
2. create/ open a screen
3. click the edit screen button on the top left of the window

**Expected behavior**
This should give all of the code for the given screen along with the ability to edit the code. Instead, you get the header, and nothing else

**Screenshots**

[image]

**Environment (please complete the following information):**
*Note: this has broken from version 6.0.1* 

- OS: [Ubuntu 20.04]
- OpenC3 COSMOS Version [e.g. 6.0.2]
- Browser: [Firefox]

jmthomas, 2025-01-21:
I tried on both Chrome and Firefox on my Mac and do not see this issue. Do you have access to another browser to confirm?

mountTom, 2025-01-21:
Previously to switch versions I : 

1. ran `./openc3.sh stop`
2. changes the version in .env
3. ran ` ./openc3.sh start`

Once I saw the following bug, I switched back to 6.0.1 and had editable access to the screens. 
To rerun on  browser (as requested) I ran ` ./openc3.sh cleanup ` between step 1 and 2. This seemed to have fixed the bug (still running in firefox). If i come accross this behavior again, I will try and delve in deeper.

Thank you for the help!

jmthomas, 2025-01-21:
Your procedures for switching versions is correct. Note that we don't guarantee going backwards in versions is going to work ... read the release notes for migrations. In general a patch release like 6.0.1 to 6.0.2 will work, while minor 6.0.2 to 6.1.0 may not, and major 5.20.0 to 6.0.0 will not.

I'll close but please re-open if you see this again. If you open your developer console in the browser you may (or may not) see Javascript errors in the console. Also check the docker logs for the cmd-tlm container: `docker logs cosmos-openc3-cosmos-cmd-tlm-api-1` and see if there are any stack traces.

closed 2025-01-21 by jmthomas