OpenC3/cosmos #1864: Is there a way to save layouts into a file?

lorenzo-gomez-windhover, 2025-01-27:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

Next search the issues here on Github. If you come up empty please submit an issue.

Is it possible to save configuration layouts into an external file(JSON, YAML, etc) and import it into COSMOS via the browser?

Or is the only way via the save/load configuration options in the browser?

Thanks
Lorenzo

jmthomas, 2025-01-28:
The standard way is to save/load configuration in the browser. However if you're in LOCAL_MODE you can get the configuration file directly in the plugins/DEFAULT/tool_config/<tool name>/config.json. This file can even be checked into CM and will be used like anything else in local mode.

I'll leave this open as a feature to allow download and upload of a config file.

ryanmelt, 2025-03-24:
Note that local mode is not currently pulling in changes for tool_configs