OpenC3/cosmos #1870: HAZARDOUS Description

lorenzo-gomez-windhover, 2025-01-31:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

Next search the issues here on Github. If you come up empty please submit an issue.

Hi there,

I have the following command definition in my ASCII file:

```
COMMAND CFS CFE_ES_Restart BIG_ENDIAN "Placeholder"
APPEND_ID_PARAMETER STREAM_ID 16 UINT 6150 6150 6150 "Packet Identification"
FORMAT_STRING "0x%04X"
APPEND_PARAMETER SEQUENCE 16 UINT MIN_UINT16 MAX_UINT16 0xC000 "Sequence count" 
APPEND_PARAMETER PKT_LEN 16 UINT 3 3 3 "Packet length" 
APPEND_PARAMETER CMD_ID 8 UINT 2 2 2 "Command code" 
APPEND_PARAMETER CHECKSUM 8 UINT MIN_UINT8 MAX_UINT8 MIN_UINT8 "" 
APPEND_PARAMETER _Payload_RestartType 16 UINT MIN_UINT16 MAX_UINT16 MAX_UINT16 "Command Error Counter:"  "LITTLE_ENDIAN" 
  STATE MAX 3
STATE POWERON 2
STATE PROCESSOR 1

SCREEN CFS cfe_es_main.txt
Hazardous "This command will SHUT DOWN flight software. Are you sure you want to send it? "
```

COSMOS does recognize the hazardous command and I get prompt every time I'm about to send it from Command Sender, or a screen.

I got the impression that the description I specify on my ASCII file ("This command will SHUT DOWN flight software. Are you sure you want to send it?") would replace the text of the default confirmation prompt. I'm assuming that's not the case, since I still get the default prompt? 

Is the description just for documentation purposes in this case?

COSMOS Version: 5.17.1

jmthomas, 2025-02-03:
The intent was that the hazardous description would show up in the prompt which is currently not happening. Thank you for bringing this to our attention.

lorenzo-gomez-windhover, 2025-02-03:
Glad I can help 👍

closed 2025-02-07 by ryan-pratt