OpenC3/cosmos #1876: OpenC3 COSMOS Init Container Connection Failure on Newer OpenC3 Versions

mattgermano, 2025-02-05:
**Describe the bug**
The OpenC3 Init container is failing to connect to the MinIO container on newer versions of OpenC3. I am seeing this on `v6.1.0`, but also on `v5.20.0`. I am attempting to upgrade from `v5.17.1`.

**To Reproduce**
Steps to reproduce the behavior:

1. `git clone -b v6.1.0 https://github.com/OpenC3/cosmos-project.git cosmos-6.1.0`
2. `git clone -b v5.17.1 https://github.com/OpenC3/cosmos-project.git cosmos-5.17.1`
3. `cd cosmos-5.17.1`
4. `./openc3.sh run`
5. View Docker logs for init container and observe success

```
❯ docker logs cosmos-5171-openc3-cosmos-init-1 -f
Wed Feb  5 15:22:58 UTC 2025
Wed Feb  5 15:22:58 UTC 2025
2025-02-05T15:22:58Z waiting for Minio http://openc3-minio:9000 RC: 0
2025-02-05T15:23:04Z waiting for Redis openc3-redis:6379. RC: 0
2025-02-05T15:23:10Z waiting for Redis Ephemeral openc3-redis-ephemeral:6380. RC: 0
...
```

6. `./openc3.sh cleanup`
7. `cd ../cosmos-6.1.0`
8. `./openc3.sh run`
9. View Docker logs for init container and observe connection failure

```
❯ docker logs cosmos-610-openc3-cosmos-init-1 -f
Wed Feb  5 15:24:15 UTC 2025
Wed Feb  5 15:24:15 UTC 2025
2025-02-05T15:24:16Z waiting for Minio http://openc3-minio:9000 RC: 6
2025-02-05T15:24:20Z waiting for Minio http://openc3-minio:9000 RC: 6
2025-02-05T15:24:26Z waiting for Minio http://openc3-minio:9000 RC: 6
```

If I `docker exec` into the init container and ping the `openc3-minio` endpoint, it returns a response so the Docker network DNS is working:

```
❯ docker exec -it cosmos-610-openc3-cosmos-init-1 sh
~ $ ping openc3-minio
PING openc3-minio (172.19.0.4): 56 data bytes
64 bytes from 172.19.0.4: seq=0 ttl=42 time=0.135 ms
64 bytes from 172.19.0.4: seq=1 ttl=42 time=0.068 ms
64 bytes from 172.19.0.4: seq=2 ttl=42 time=0.084 ms
64 bytes from 172.19.0.4: seq=3 ttl=42 time=0.058 ms
64 bytes from 172.19.0.4: seq=4 ttl=42 time=0.068 ms
```

However, attempting to `curl` the endpoint results in a failure:

```
~ $ curl http://openc3-minio:9000
curl: (6) Could not resolve host: openc3-minio
```

If I run `curl` in the `v5.17.1` Init container, it successfully returns a response:

```
~ $ curl http://openc3-minio:9000
<?xml version="1.0" encoding="UTF-8"?>
<Error><Code>AccessDenied</Code><Message>Access Denied.</Message>
```

**Expected behavior**
OpenC3 init connection success during startup.

**Environment (please complete the following information):**

- OS: Ubuntu 22.04
- OpenC3 COSMOS Version: `v6.1.0`
- Browser: Chrome
- Docker Compose Version:  `v2.32.4`
- Docker Version: `v27.5.1`

jmthomas, 2025-02-05:
I just ran your exact instructions and it completed successfully and I was able to curl minio (with access denied). Note that by doing `cleanup` in step 6 you're not 'upgrading' but actually deleting the previous version and starting over with 6.1.0. This also works for me.

I did the upgrade process from 6.0.2 to 6.1.0 successfully in the cosmos-project by stopping the containers, switching to the new v6.1.0 tag, and then starting the containers.

It doesn't make sense that you would be able to ping minio but not curl. The [return code](https://everything.curl.dev/cmdline/exitcode.html) of 6 means "Could not resolve host" but then how could ping work?

I'm on Mac OS with Docker Desktop v4.37.2 (docker v27.4.0) and docker compose v2.31.0. So we're close but not exact but this doesn't sound like a docker / compose issue to me.

ryanmelt, 2025-02-05:
Definitely some sort of ipv6 issue.   We could force curl to use ipv4, but I would like to understand why your configuration doesn't like ipv6

https://serverfault.com/questions/335359/how-is-it-possible-that-i-can-do-a-host-lookup-but-not-a-curl

ryanmelt, 2025-02-05:
Can you look at what DNS server your host is configured for?  That might be the issue

mattgermano, 2025-02-05:
@jmthomas @ryanmelt I was able to fix it.

I exec'd into the container again and started commenting stuff out in `/etc/resolv.conf` and retrying `curl` until it started working. It turns out that I had an old/expired search domain set on my Ethernet interface. When I commented out the `search` line in `/etc/resolv.conf` the `curl` immediately worked. I found which domain was bad, removed it from the Ethernet interface, restarted the interface, and it's now working correctly.

Thank you very much for the help in narrowing that down. That would have taken me a while to figure out.

closed 2025-02-05 by mattgermano