OpenC3/cosmos #1895: Parameter default value does not work for hexadecimal values

timothyscherer, 2025-02-12:
I have a command packet definition:
```log
COMMAND MC NOOP BIG_ENDIAN "..."
  APPEND_ID_PARAMETER CCSDS_STREAMID 16 UINT MIN MAX 0x1834 "CCSDS Packet Identification"
  APPEND_PARAMETER CCSDS_SEQUENCE    16 UINT MIN MAX 0xC000 "CCSDS Packet Sequence Control"
  APPEND_PARAMETER CCSDS_LENGTH      16 UINT MIN MAX 1 "CCSDS Packet Data Length"
  APPEND_PARAMETER CCSDS_FC           8 UINT MIN MAX 0 "CCSDS Command Function Code"
  APPEND_PARAMETER CCSDS_CHECKSUM     8 UINT MIN MAX 0 "Checksum"
```

The parameter CCSDS_STREAMID has a specified default of 0x1834. This is parsed out as a string, and in [packet_item.py:620](https://github.com/OpenC3/cosmos/blob/main/openc3/python/openc3/packets/packet_item.py#L620), this string is converted to an integer. This conversion fails due to assumed base 10:

```py
Exception: Error processing MC:
Traceback (most recent call last):
  File "/openc3/venv/lib/python3.11/site-packages/openc3/packets/packet_item.py", line 620, in convert
    return int(value)
           ^^^^^^^^^^
ValueError: invalid literal for int() with base 10: '0x1834'
```

Solution (?):
Replace
```py
return int(value)
```
with
```py
return int(value, 0)
```
Specifying the base of 0 is required in order to [decode strings with hex prefixes](https://docs.python.org/3/library/functions.html#int)

I am unsure why it is using the Python code to do this - I tested on 5.16.2 AND 5.20.1 and 6.1.0 and got the same results. Is there a way to use the ruby code for this instead as a temporary fix?

jmthomas, 2025-02-12:
This appears to be happening just for ID items. I can confirm this is a bug.

Your target must be a python target declared with `LANGUAGE python` for the python code to be parsing it.

timothyscherer, 2025-02-12:
Where would that be defined? In the target file itself? Is that the default if nothing is specified?

ryanmelt, 2025-02-12:
Its defined in the target.txt file.
Ruby is the default.

timothyscherer, 2025-02-12:
I do not have that specified anywhere in the entire plugin. Is it possible that it uses that because the plugin contains a `requirements.txt` (although it is empty)

timothyscherer, 2025-02-12:
The gemspec also has `s.platform = Gem::Platform::RUBY`, but that is the only python/ruby related thing I see in the entire plugin

ryanmelt, 2025-02-12:
If you have a python based interface, it will also be processed by Python (Is that that case?).
Target code/configuration only executes in interfaces/routers/ and the decom microservice.

ryanmelt, 2025-02-12:
> The gemspec also has `s.platform = Gem::Platform::RUBY`, but that is the only python/ruby related thing I see in the entire plugin

The gemspec will always say Ruby because its a ruby gem.  That doesn't affect whether or not the contents are ruby or python.

timothyscherer, 2025-02-12:
I did have a python based interface - I changed it back and it is happy now. Thank you!

PR #1896 (2025-02-13): Properly parse hex and binary ID items
closes #1895
closed 2025-02-13 by jmthomas