OpenC3/cosmos #1908: Failed Docker instructions to remove and clean up Gems on openc3-base image

NuclearFizzler, 2025-02-21:
I'm building a COSMOS gem via ```openc3.sh cli rake build VERSION=(number)```, and I've been getting the following warning. The gem still builds fine, that's why this is a question and not a bug report. I decided to track down the source of the issue.

```
WARN: Unresolved or ambiguous specs during Gem::Specification.reset:
      stringio (>= 0)
      Available/installed versions of this gem:
      - 3.1.2
      - 3.0.4
WARN: Clearing out unresolved specs. Try 'gem cleanup <gem>'
```
From some digging, stringio 3.1.2 is coming from the Gemfile.lock inside the image, while I suspect 3.0.4 is coming from installing RubyGems 3.3.14 based on the release dates of both versions (May and December 2022). I know that migrations are in progress; however, why is RubyGems 3.3.14 required? 

Additionally, it seems that the instructions to remove grpc and google-protobuf did not run successfully. I opened a terminal on the latest version of _cmd-tlm-api_ and found the following:

[image]

Running the uninstall command with the ```--debug``` flag revealed issues with the flags set in gemrc.

[image]

I'm not sure what the overall impact of these issues are. I'd like to hear your feedback on these issues, given that you have more experience working on COSMOS.

NuclearFizzler, 2025-02-21:
I did some more testing with the docker images; cleaning up the gem options cleared some of the exceptions, but the uninstall commands still seem to have no effect. The gem still appears in ```gem list``` afterward.

jmthomas, 2025-02-21:
The grpc stuff is a little weird because we need to install with the `--platform ruby` flag and then uninstall the `--platform aarch64-linux` flag. I think you're on to something with the .gemrc stuff though. I'm doing some local testing and will let you know.

jmthomas, 2025-02-21:
Thanks @NuclearFizzler!

closed 2025-02-21 by jmthomas

jrericks, 2025-04-03:
The "Unresolved or ambiguous specs during Gem::Specification.reset" warning is still present after PR 1909. See UT output from this PR: https://github.com/OpenC3/cosmos/actions/runs/13453803311/job/37609348991#step:3:37. Also we see this warning when building gems with openc3 6.2.1.