OpenC3/cosmos #1912: LineGraph loses items after setting custom scaling

NuclearFizzler, 2025-02-22:
**Description**
After setting custom min/max Y-values to override automatic scaling, the graphed items disappear from the graph.

**To Reproduce**
Steps to reproduce the behavior:

1. Right click on an active graph in Telemetry Viewer and select 'Edit Graph (id)'
2. Click on 'Scale / Lines' and set a value in the 'Min Y' or 'Max Y Axis' fields
3. Select OK
4. Items disappear from the graph
5. Verify by going to 'Edit Graph' > 'Items;' no items appear in the list
6. Related error appears in Developer Console

**Expected behavior**
Graph items should remain on the graph after custom scaling is set.

**Screenshots**
_Sample graph from the Demo_
[image]

_Scaling set in Edit Graph menu_
[image]

_Items disappear_
[image]

_Error in Developer Console_
[image]

**Environment (please complete the following information):**

- OS: Windows 11 Docker Desktop with WSL2 integration
- OpenC3 COSMOS Version: latest
- Browser: Firefox

NuclearFizzler, 2025-02-23:
What tools does OpenC3 use to debug Vue? I've been trying to set up Vue DevTools via my browser, but attempting to run it in COSMOS keeps returning "Devtools inspection not available." I've tried modifying the Vite config to allow devtools, but there seems to be no effect.

jmthomas, 2025-02-23:
We serve a local copy of the tool in question by following this guide: https://docs.openc3.com/docs/development/developing#running-a-frontend-application. Then we just use the local console and dev tools. Note that you should also run a `yarn serve` in the openc3-vue-common directory if you're using any of those files (you probably are).

PR #1979 (2025-03-31): Fix Graph without overview
closes #1912
closed 2025-03-31 by jmthomas