OpenC3/cosmos #1913: Development Documentation Help

NuclearFizzler, 2025-02-22:
My teammates and I are attempting to implement #1710 and investigate #1912, and we've found some outdated information in the development documentation located at <https://docs.openc3.com/docs/development/developing>.

The commit implementing anycable removed the ```dev``` option from openc3.sh/bat. I assume we can just run COSMOS with ```openc3.sh start```, but we could like to see this reflected in the docs.

I'm still trying to wrangle together the yarn environment. I'll add additional updates if we note anything.

NuclearFizzler, 2025-02-22:
I was just now able to reverse-engineer the process of how to setup the frontend to start developing with. Here are the needed updates to the documentation.

1) To bootstrap the frontend, you must run yarn not in the `openc3-cosmos-init` folder, but in the `init/plugin` folder. 
2) You must run `yarn` and `yarn build:common` to build the plugins correctly.
3) Running `yarn serve` no longer produces output showing the URL of the built webpages; I understand this is a holdover from 7a22ffaca09d78162137afd002086b79c82c0182 since we are running `build` and not `preview`

jmthomas, 2025-02-22:
Thanks @NuclearFizzler. We will update https://docs.openc3.com/docs/development/developing to make these changes.

PR #1917 (2025-02-23): Dev doc updates
closes #1913
closed 2025-02-23 by jmthomas