OpenC3/cosmos #1922: Differently Formatted Dates for the Same Item when Using Uppercase Versus Lowercase

stephen-ritter, 2025-02-26:
The following is an example screen showing the issue. It appears that dates are being formatted differently when depending on whether you are using and uppercase version of the string versus a lowercase version of the string.
```
SCREEN AUTO AUTO 1.0

VERTICAL
  LABELVALUE INST ADCS PACKET_TIMEFORMATTED CONVERTED 24
  LABELVALUE INST ADCS packet_timeformatted CONVERTED 24
  HORIZONTALLINE
  LABELVALUE INST ADCS RECEIVED_TIMEFORMATTED CONVERTED 24
  LABELVALUE INST ADCS received_timeformatted CONVERTED 24
  HORIZONTALLINE
  LABELVALUE INST ADCS BIASX CONVERTED 24
  LABELVALUE INST ADCS biasx CONVERTED 24
END
```

[image]

ryanmelt, 2025-02-26:
Interesting catch.  The frontend applies local time / utc time automatically to certain items.   It must be case sensitive.

jmthomas, 2025-02-27:
Found in `VWidget.js` line 249, 250 but we should make sure we're not assuming all caps in other places. Specifically settings in `Widget.js`.

closed 2025-02-27 by ryan-pratt