OpenC3/cosmos #1942: Write Conversion for String to UINT16/UINT32, Implicit Conversion Error

ldurb, 2025-03-05:
I am not sure if my issue is solved by #1825 , but I am attempting to use the Command Sender to input a parameter and then apply my write_conversion below.
`

        def call(value,packet,buffer)

            int_value = value.to_i

            return 0 if int_value==0

            bitmask = (1 << (@int_bits + @fract_bits)) - 1
            int_value = (int_value * (2 ** @fract_bits)).to_i.round

            int_value &= bitmask

            return int_value 
       end
   

` 
The parameter and write_conversion are written as: 

`PARAMETER        TEST_PARAMETER   0             16              UINT        0               1                 ".25"
  WRITE_CONVERSION fixed_point_conversion_write.rb  1   15 `

When I attempt to send the command using the default value, I get an  " RuntimeError: no implicit conversion of String into Integer" 

What am I missing? I believe I followed the documentation which suggests setting the default value in the situation in String format for a value that will ultimately end up as UINT, hence the "0.25"

Note: If I disable parameter conversions, I do not get this error. 

Please help!

jmthomas, 2025-03-05:
I think that it is resolved in COSMOS 6.2.0. Please try and let me know if this is resolved.

ldurb, 2025-03-05:
I have pulled a fresh repo with COSMOS 6.2.0 and still get the same error : " RuntimeError: no implicit conversion of String into Integer"

ryanmelt, 2025-03-05:
Try "0.25"

ldurb, 2025-03-05:
Yes I have that set as my default value, quotes included, I fixed the code snippet above for clarity.

ldurb, 2025-03-05:
After adjusting the default value and also attempting to input a new value using Command Sender, I still receive the same error.

jmthomas, 2025-03-09:
I added the following to the demo's inst_cmd.txt file: 

```
COMMAND INST QUIET BIG_ENDIAN "Enable/disable no out of limits in the demo"
  VALIDATOR inst_cmd_validator.rb
  <%= render "_ccsds_cmd.txt", locals: {id: 11} %>
  APPEND_PARAMETER STATE 8 UINT 0 1 1
    STATE FALSE 0
    STATE TRUE 1
  APPEND_PARAMETER TEST_PARAMETER 16 UINT 0 1 "0.25" # <= NEW
    WRITE_CONVERSION fixed_point_conversion_write.rb 1 15   # <= NEW
```

And then implemented `INST/lib/fixed_point_conversion_write.rb` as follows:
```ruby
require 'openc3/conversions/conversion'
module OpenC3
  class FixedPointConversionWrite < Conversion
    def initialize(int_bits, fract_bits)
      @int_bits = int_bits
      @fract_bits = fract_bits
      @converted_type = :UINT
      @converted_bit_size = 16
      @converted_array_size = nil
    end
    def call(value, _packet, _buffer)
      int_value = value.to_i
      return 0 if int_value==0
      bitmask = (1 << (@int_bits + @fract_bits)) - 1
      int_value = (int_value * (2 ** @fract_bits)).to_i.round
      int_value &= bitmask
      return int_value
    end
  end
end
```
The command was successfully sent but when I viewed the raw data it was `00 00` because any fractional value when you call .to_i will be 0.

ldurb, 2025-03-12:
I found out the issue had to do with the initializing. I never converted the int_bits, fract_bits variables from String to Integer.  Thank you for the help!

closed 2025-03-12 by ldurb