OpenC3/cosmos #1944: cli Migrate no longer functioning

mountTom, 2025-03-05:
**Describe the bug**
In version 6.2.0, try to call the `C:\COSMOS> C:\cosmos-project\openc3.bat cli migrate -a demo` described in [https://docs.openc3.com/docs/getting-started/upgrading](https://docs.openc3.com/docs/getting-started/upgrading)
 
**To Reproduce**
Steps to reproduce the behavior:

1. try to migrate a COSMOS 4 target with COSMOS 6
2. `Unknown task: migrate`

**Expected behavior**
I would expect the migration to still work in cosmos 6. When I revert to the last COSMOS 5 version, it works.

**Environment (please complete the following information):**

- OS: [Ubuntu 20.04]
- OpenC3 COSMOS Version [e.g. 6.2.0]
- Browser: [Firefox]

jmthomas, 2025-03-05:
We explicitly removed this capability in COSMOS 6 here: https://github.com/OpenC3/cosmos/pull/1659. Is this something you need for a program? I could see restoring it ... I don't think that it saved us any other dependencies. @ryanmelt?

mountTom, 2025-03-06:
I just needed to port over some Cosmos 4 code. Switching to Cosmos 5 worked fine, just got a little confused with the documentation

jmthomas, 2025-03-22:
I think we'll leave the migration tool as a COSMOS 5 capability

closed 2025-03-22 by jmthomas