OpenC3/cosmos #1947: protocol_cmd cannot update instance variable

tsjordan-eng, 2025-03-06:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Describe the bug**
I am trying to use a `protocol_cmd` to change an instance variable (`@var`) at runtime so that the behavior of `write_data` which uses `@var` changes. However, though I can see from log messages that `@var` has changed, that change does not happen in `write_data`. Is `write_data` in a different thread or instance of the protocol? Is there a better way to change realtime behavior than to set an intance variable?

**To Reproduce**
Steps to reproduce the behavior:

1. Make a custom protocol with a `protocol_cmd` and a variable `@var`
2. create a log message in `protocol_cmd` that proves you're setting `@var`
3. create a log message in `write_data` that shows `@var` is unchanged.

**Expected behavior**
`@var` should be updated.

**Environment (please complete the following information):**

- OS: RHEL8
- OpenC3 COSMOS Version 5.20.0
- Browser: FireFox

jmthomas, 2025-03-08:
I created a protocol called `MyProtocol` and put it in the Demo's `EXAMPLE/lib/my_protocol.rb`:

```
require 'openc3/interfaces/protocols/protocol'

module OpenC3
  class MyProtocol < Protocol
    def initialize(allow_empty_data = nil)
      super(allow_empty_data)
      @var = nil
      @args = nil
    end
    def write_data(data, extra = nil)
      print("cmd:#{@var} args:#{@args}")
      return super(data, extra)
    end
    def protocol_cmd(cmd_name, *cmd_args)
      @var = cmd_name
      @args = cmd_args
    end
  end
end
```

I then added it to the Example interface in the plugin.txt:

```
  INTERFACE <%= example_int_name %> example_interface.rb <%= example_host %> <%= example_port %>
    MAP_TARGET <%= example_target_name %>
    DONT_CONNECT
    # Override the default log time of 600
    LOG_STREAM 60
    OPTION CONNECT_CMD LOG "<%= example_target_name %> START"
    PROTOCOL WRITE MyProtocol
```

Initially this prints `cmd: args:` when you send a command. I then sent the protocol command by running this script in Script Runner: `interface_protocol_cmd("EXAMPLE_INT", "HI", "THERE")`

At that point when I send the `EXAMPLE START` command I see the following in the docker logs:
```
cmd:HI args:["THERE"]{"time":1741460025329454251,"@timestamp":"2025-03-08T18:53:45.329454Z","level":"INFO","microservice_name":"DEFAULT__INTERFACE__EXAMPLE_INT","user":"anonymous","container_name":"8819ac4c399b","message":"cmd(\"EXAMPLE START\")","type":"log"}
```

So I'm definitely affecting these variables. How are you sending your protocol command?

tsjordan-eng, 2025-03-18:
> @tsjordan-eng <https://github.com/tsjordan-eng> any luck?

Sorry I missed your response! I'll try installing your microservice and
compare it to mine and get right back to you.

jmthomas, 2025-04-11:
Closing due to lack of info. We can reopen if you can reproduce in our Demo.

closed 2025-04-11 by jmthomas