OpenC3/cosmos #1958: Running COSMOS from a subpath

0lionelzhang0, 2025-03-20:
Hello,

Is there a way to configure COSMOS to run on a subpath, e.g. `http://localhost/cosmos` instead of `http://localhost:2900`?

Thanks,
Lionel

ryanmelt, 2025-03-24:
I haven't tried this but it should be possible.  Would require modifying our traefik.yaml file to strip off the extra subpath before passing urls on to the internal services.

0lionelzhang0, 2025-03-25:
I think modifying traefik.yaml might not be sufficient since all the Tool hyperlinks within the webapp also need the subpath, e.g. http://localhost:2900/tools/cmdtlmserver -> http://localhost/cosmos/tools/cmdtlmserver

Is there a way to update those as well through some config/env value globally?

ryanmelt, 2025-03-25:
Ah yes, this will probably fall apart with the tools.   The path part of the url is built into the tools, so they are not going to like having anything additional there.

Could you use a subdomain rather than a subpath?

0lionelzhang0, 2025-03-26:
Got it, mostly wanted to double check I wasn't missing anything in the docs for setting subpath up.

I have been experimenting with subdomain on localhost e.g. `cosmos.localhost`, which seems to work fine. I'll need to reach out to our IT on setting up DNS for servers. Thanks!

closed 2025-07-09 by jmthomas