OpenC3/cosmos #1960: Load utility side effect clobbers instrumented variables/types/classes that share a name

alex-pope1901, 2025-03-21:
**Describe the bug**
The issue I originally ran into was that I was running a python script which had an enum that was specific to the script. This script calls load_utility on another script that is related and had an enum of the same name. This ended up causing my pydantic model to throw an error since it said that the expected type of the original enum was not the same as the enum from the sub script. This is avoidable but load_utility should not change how python variable scope works.

**To Reproduce**
Create the two scripts in scriptrunner. First called test_1.py with the following contents:
```python
class Node:
  
  @staticmethod
  def print_msg():
    print("This is the node I'm expecting.")
    
Node.print_msg()
load_utility("LDPE3A/test_2.py")
Node.print_msg()
```

Second make a script called test_2.py with the following contents:
```python
class Node:
  
  @staticmethod
  def print_msg():
    print("Hey this isn't the original node.")
```

Run the test_1.py and verify that the output looks like:

```
(test_1.py:5): This is the node I'm expecting.
(test_2.py:5): Hey this isn't the original node.
(SCRIPTRUNNER): Script completed: test_1.py
```

This shows that the class is replaced in the parent script by the child script after load_utility is called on the child. Note that the logs show the code executing from test_2 instead of test_1.

**Expected behavior**
Scope should not be modified by load_utility. If a class with the same name is defined in a sub script it should not replace it in the parent with load_utility. In the example, this means that the prints before and after load_utility should look the same. Unintended side effects like this make instrumenting code unreliable.

**Screenshots**

[image]
test_1.py code

[image]
test_2.py code

[image]
test_1.py output

**Environment (please complete the following information):**

- OS: Windows 11
- OpenC3 COSMOS Enterprise Version 6.1.0
- Browser: Firefox

jmthomas, 2025-03-22:
`load_utility` is effectively doing a python import. So if you change your file to this:

```
class Node:

  @staticmethod
  def print_msg():
    print("This is the node I'm expecting.")

Node.print_msg()
from test2 import Node
Node.print_msg()
```

And keep the other one the same you'll see the default behavior in Python is the same:

```
% python3 test1.py
This is the node I'm expecting.
Hey this isn't the original node.
```

jmthomas, 2025-03-24:
We actually load the file and then `exec` it in the current context but that's going to overwrite any duplicate classes or methods  defined in the current context. To me this is the desired and expected behavior.

alex-pope1901, 2025-03-25:
Thank you for the clarification. It would definitely be helpful to have more control over instrumenting scripts. There are times I want to have operator visibility into the sub-procedure but using load_utility will clobber a class or enum I have in the parent. I appreciate the support!