OpenC3/cosmos #1968: Recommended configuration for multiple cmd interfaces to a target?

0lionelzhang0, 2025-03-25:
Hello,

We have a "wired" and "wireless" interface mapped to our target. It is pretty common for us to want to test telemetry/commanding over one interface or the other depending on the test. However, when one of the interfaces is DISCONNECTED, commanding fails with the error "Interface not connected". Is this the correct usage of interfaces? It would be handy if there was an interface modifier to specify it is not required for commanding, for example.

I found this old thread about the same topic and was wondering if using map_target_to_interface is still the recommended approach? 
https://github.com/OpenC3/cosmos/issues/595

Thanks!
Lionel

jmthomas, 2025-03-25:
Interfaces must be connected to be commanded. If you want to switch your target's commanding from one interface to anther you should use the `map_target_to_interface` routine. Your link to that other issue is a good place to start.

closed 2025-03-25 by jmthomas

0lionelzhang0, 2025-03-25:
Got it, thanks!