OpenC3/cosmos #1974: display wait condition and timeout

321github123, 2025-03-27:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Is your feature request related to a problem? Please describe.**
Our customer would like to see what condition we are waiting on and what the time out value is

**Describe the solution you'd like**
wait condition and time out are displayed in script runner 

[image]

**Additional context**
Add any other context or screenshots about the feature request here.

jmthomas, 2025-03-27:
The condition waited on is visible in the script being executed. I'm confused by the request.

ryanmelt, 2025-03-27:
I'm also confused.  The condition and timeout are both in the highlighted wait line.

321github123, 2025-03-27:
Sorry for the confusion

We would like to know what the "condition" and "timeout" are.  In many of our scripts, those values are passed in as variables.

[image]

ryanmelt, 2025-03-27:
When could potentially print that into the script log.

321github123, 2025-03-27:
that would be helpful

jmthomas, 2025-12-23:
I think this is resolved. The following script:

```python
wait("INST HEALTH_STATUS TEMP1 > 0", 5)
check("INST HEALTH_STATUS TEMP1 > 0")
wait_check("INST HEALTH_STATUS TEMP1 > 0", 5)
```

Produces this result:

```
2025-12-23T15:35:49.247988Z (SCRIPTRUNNER): Starting script: 2025_12_23_08_35_48_653_temp.py, line_delay = 0.1
--
2025-12-23T15:35:49.248305Z (2025_12_23_08_35_48_653_temp.py:0): 2025-12-23T15:35:49.247988Z (SCRIPTRUNNER): Starting script: 2025_12_23_08_35_48_653_temp.py, line_delay = 0.1
2025-12-23T15:35:49.357777Z (2025_12_23_08_35_48_653_temp.py:1): WAIT: INST HEALTH_STATUS TEMP1 > 0 success with value == 29.204100000000007 after waiting 0.006 seconds
2025-12-23T15:35:49.468052Z (2025_12_23_08_35_48_653_temp.py:2): CHECK: INST HEALTH_STATUS TEMP1 > 0 success with value == 29.204100000000007
2025-12-23T15:35:49.573144Z (2025_12_23_08_35_48_653_temp.py:3): CHECK: INST HEALTH_STATUS TEMP1 > 0 success with value == 29.204100000000007 after waiting 0.004 seconds
2025-12-23T15:35:49.675920Z (SCRIPTRUNNER): Script completed: __TEMP__/2025_12_23_08_35_48_653_temp.py
```

So the condition is printed in all cases.

closed 2025-12-23 by jmthomas