OpenC3/cosmos #1977: Back-orbit data ingestion

0lionelzhang0, 2025-03-28:
Hello,

What is the best practice for ingesting back-orbit or historical telemetry into COSMOS? For example, we downlink some data files containing back-orbit data from vehicle to ground, should we just send that data packet-by-packet into the target interface? Or is there a more efficient way to backfill data?

Each packet has a "generation time" tlm value that we map to PACKET_TIME.

Thanks,
Lionel

ryanmelt, 2025-03-28:
If you are downlinking files you can use the FileInterface.  This is the only included Interface we haven't documented yet but you can find the source here: https://github.com/OpenC3/cosmos/blob/main/openc3/python/openc3/interfaces/file_interface.py

Currently would be reading files from a mounted volume.  It would be pretty easy to make an equivalent BucketInterface if the files are stored in Buckets.

0lionelzhang0, 2025-03-28:
Thanks, Ryan. This looks like it'd be perfect for our case.

closed 2025-03-28 by 0lionelzhang0