OpenC3/cosmos #1978: Build Failure of Generated Plugin Due to Missing `yarn` and `vite` in openc3-cosmos-cmd-tlm-api Container

jurgensymynck, 2025-03-29:
Hi,

I am attempting to build a HelloworldWidget, following [these guidelines](https://docs.openc3.com/docs/guides/custom-widgets#custom-widgets), and the [code generator documentation](https://docs.openc3.com/docs/getting-started/generators).

Generating a plugin, target, conversion and HelloworldWidget work fine.
When trying to build my plugin however, `openc3.bat` (I develop on Windows) throws this error:

```
openc3.bat cli rake --verbose --trace build VERSION=1.0.0-Alpha

** Invoke build (first_time)
** Invoke require_version (first_time)
** Execute require_version
** Execute build
yarn run build
rake aborted!
Command failed with status (127): [yarn run build]
```
This indicates than `yarn` is not available in the container `openc3-cosmos-cmd-tlm-api`.
Neither is `vite`, and after some AI-assisted trial and error, a solution was found: having `yarn` and `vite` installed on the fly when the container is created.

There are some dependency problems that seem to be addressable with `--legacy-peer-deps`

I have the same problems (briefly tested) using the `cosmos` repo instead of the recommended `cosmos-project` repo.

I cannot imagine that these undocumented steps are the way to go...
It feels that the code generation functionality was based on a less minimalistic linux docker image than the current Alpine images...
What am I missing, and is there a better way?

greetings, Jurgen

---

### Relevant info

- Windows 11
- up to date clone of `cosmos-project` repo (also tested briefly with `cosmos` repo)
- cosmos 6.2.1

### dependency error

```
npm install --save-dev vite --cache /tmp/.npm-cache
npm ERR! code ERESOLVE
npm ERR! ERESOLVE unable to resolve dependency tree
npm ERR!
npm ERR! While resolving: HallowereldWidget@6.2.1
npm ERR! Found: vite@5.4.15
npm ERR! node_modules/vite
npm ERR!   dev vite@"^5.4.11" from the root project
npm ERR!
npm ERR! Could not resolve dependency:
npm ERR! peer vite@"^2.0.0" from vite-plugin-style-inject@0.0.1
npm ERR! node_modules/vite-plugin-style-inject
npm ERR!   dev vite-plugin-style-inject@"^0.0.1" from the root project
```

## new file `Dockerfile.openc3-yarn`

```
ARG BASE_IMAGE
FROM ${BASE_IMAGE}

USER root

RUN apk add --no-cache nodejs npm \
 && npm install -g yarn vite

USER ${OPENC3_USER_ID:-1001}
```

## `compose.yaml`
```
...
openc3-cosmos-cmd-tlm-api:
    user: "${OPENC3_USER_ID:-1001}:${OPENC3_GROUP_ID:-1001}"
    # image: "${OPENC3_REGISTRY}/${OPENC3_NAMESPACE}/openc3-cosmos-cmd-tlm-api${OPENC3_IMAGE_SUFFIX}:${OPENC3_TAG}"
    build:
      context: .
      dockerfile: Dockerfile.openc3-yarn
      args:
        BASE_IMAGE: "${OPENC3_REGISTRY}/${OPENC3_NAMESPACE}/openc3-cosmos-cmd-tlm-api${OPENC3_IMAGE_SUFFIX}:${OPENC3_TAG}"
...
```

## `Rakefile`
```
PLUGIN_NAME = Dir['*.gemspec'][0].split('.')[0..-2].join('.')

task :require_version do
  unless ENV['VERSION']
    puts "VERSION is required: rake build VERSION=X.Y.Z"
    exit 1
  end
end

task :build => [:require_version] do
  _, platform, *_ = RUBY_PLATFORM.split("-")
  if platform == 'mswin32' or platform == 'mingw32'
    puts "Warning: Building gem on Windows will lose file permissions"
  end

  # Use temporary cache to avoid EACCES 
  npm_cache = '/tmp/.npm-cache'

  # Install vite locally if not already present 
  unless File.exist?("node_modules/vite")
    sh('npm', 'install', '--save-dev', 'vite', '--cache', '/tmp/.npm-cache', '--legacy-peer-deps')
  end

  sh('yarn', '--verbose', 'run', 'build')
  sh('gem', 'build', PLUGIN_NAME)
  sh('openc3cli validate *.gem') do |ok, status|
    if !ok && status.exitstatus == 127
      puts "Install the openc3 gem to validate! (gem install openc3)"
    end
  end
end
```

jurgensymynck, 2025-03-31:
UPDATE:

It seems my above post is the result of two crippled approaches to solve my initial `yarn run build failed` error after running `openc3.bat cli rake --verbose --trace build VERSION=1.0.0-Alpha` in my plugin folder.

I will proceed debugging this and report back.
I will remove the confusing settings in the previous post and keep only the relevant stuff.

greetings, Jurgen

closed 2025-07-09 by jmthomas

rgmann-bae, 2025-07-28:
@jurgensymynck @jmthomas Have you identified a solution to this? I'm running into the same problem. I've also started the `openc3-cosmos-cmd-tlm-api` container interactively and it doesn't appear to have yarn or vite installed.

**Steps to reproduce**

1. `./openc3.sh cli generate plugin test-widgets`
2. `cd ./test-widgets`
3. `../openc3.sh cli generate widget HelloworldWidget`
4. `../openc3.sh cli rake build VERSION=0.0.1 --trace`

**Output**

```
** Invoke build (first_time)
** Invoke require_version (first_time)
** Execute require_version
** Execute build
yarn run build
rake aborted!
Command failed with status (127): [yarn run build]
/usr/lib/ruby/gems/3.3.0/gems/rake-13.3.0/lib/rake/file_utils.rb:65:in `block in create_shell_runner'
/usr/lib/ruby/gems/3.3.0/gems/rake-13.3.0/lib/rake/file_utils.rb:57:in `sh'
/openc3/local/Rakefile:17:in `block in <top (required)>'
/usr/lib/ruby/gems/3.3.0/gems/rake-13.3.0/lib/rake/task.rb:281:in `block in execute'
/usr/lib/ruby/gems/3.3.0/gems/rake-13.3.0/lib/rake/task.rb:281:in `each'
/usr/lib/ruby/gems/3.3.0/gems/rake-13.3.0/lib/rake/task.rb:281:in `execute'
/usr/lib/ruby/gems/3.3.0/gems/rake-13.3.0/lib/rake/task.rb:219:in `block in invoke_with_call_chain'
/usr/lib/ruby/gems/3.3.0/gems/rake-13.3.0/lib/rake/task.rb:199:in `synchronize'
/usr/lib/ruby/gems/3.3.0/gems/rake-13.3.0/lib/rake/task.rb:199:in `invoke_with_call_chain'
/usr/lib/ruby/gems/3.3.0/gems/rake-13.3.0/lib/rake/task.rb:188:in `invoke'
/usr/lib/ruby/gems/3.3.0/gems/rake-13.3.0/lib/rake/application.rb:188:in `invoke_task'
/usr/lib/ruby/gems/3.3.0/gems/rake-13.3.0/lib/rake/application.rb:138:in `block (2 levels) in top_level'
/usr/lib/ruby/gems/3.3.0/gems/rake-13.3.0/lib/rake/application.rb:138:in `each'
/usr/lib/ruby/gems/3.3.0/gems/rake-13.3.0/lib/rake/application.rb:138:in `block in top_level'
/usr/lib/ruby/gems/3.3.0/gems/rake-13.3.0/lib/rake/application.rb:147:in `run_with_threads'
/usr/lib/ruby/gems/3.3.0/gems/rake-13.3.0/lib/rake/application.rb:132:in `top_level'
/usr/lib/ruby/gems/3.3.0/gems/rake-13.3.0/lib/rake/application.rb:83:in `block in run'
/usr/lib/ruby/gems/3.3.0/gems/rake-13.3.0/lib/rake/application.rb:214:in `standard_exception_handling'
/usr/lib/ruby/gems/3.3.0/gems/rake-13.3.0/lib/rake/application.rb:80:in `run'
/usr/lib/ruby/gems/3.3.0/gems/rake-13.3.0/exe/rake:27:in `<top (required)>'
/usr/bin/rake:25:in `load'
/usr/bin/rake:25:in `<main>'
Tasks: TOP => build
```

**Relevant info**
Host: RHEL 8
cosmos 6.6.0