OpenC3/cosmos #1992: Tokenized search for dropdowns

0lionelzhang0, 2025-04-01:
**Is your feature request related to a problem? Please describe.**
For longer packet or item names, it is frustrating to have to type string that matches exactly to the thing you're looking for. 

**Describe the solution you'd like**
COSMOS currently allows for autocomplete in the Target, Packet and Item selects. However, it is limited to a single search string that needs to match exactly. This enhancement would allow for more flexible and user-friendly search, e.g. search `app something` -> `MY_APPLICATION/COMMAND_FOR_SOMETHING`.

0lionelzhang0, 2025-04-01:
Yes I think that ticket + this tokenized/fuzzy search would make the autocomplete much more user-friendly.

jmthomas, 2026-09-01:
Either a regex search or a split token search like `HEALTH STAT` which would match `HEALTH_STATUS`.

closed 2026-10-13 by calmonroe