OpenC3/cosmos #2006: COSMOS 6.0.2 not seeing built in packet_time_formatted

badrobit, 2025-04-07:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Describe the bug**
The plugin build passes but deploy fails due to it not knowing how to find the `packet_time_formatted_conversion.py`. 

```bash
Traceback (most recent call last):
  File \"/openc3/venv/lib/python3.11/site-packages/openc3/packets/packet_config.py\", line 584, in process_current_item
    klass = get_class_from_module(
            ^^^^^^^^^^^^^^^^^^^^^^
  File \"/openc3/venv/lib/python3.11/site-packages/openc3/top_level.py\", line 158, in get_class_from_module
    importlib.import_module(f\".{parts[-1]}\", \".\".join(parts[0:-1]))
  File \"/usr/lib/python3.11/importlib/__init__.py\", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File \"<frozen importlib._bootstrap>\", line 1204, in _gcd_import
  File \"<frozen importlib._bootstrap>\", line 1176, in _find_and_load
  File \"<frozen importlib._bootstrap>\", line 1140, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'openc3.conversions.packet_time_conversion'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File \"/openc3/venv/lib/python3.11/site-packages/openc3/system/system.py\", line 124, in add_target
    self.packet_config.process_file(cmd_tlm_file, target.name)
  File \"/openc3/venv/lib/python3.11/site-packages/openc3/packets/packet_config.py\", line 255, in process_file
    self.process_current_item(parser, keyword, params)
  File \"/openc3/venv/lib/python3.11/site-packages/openc3/packets/packet_config.py\", line 598, in process_current_item
    raise parser.error(error)
openc3.config.config_parser.ConfigParser.Error: No module named 'openc3.conversions.packet_time_conversion'

```

**To Reproduce**
Steps to reproduce the behavior:

1. Create a plugin that requires the [`packet_time_formatted`](https://github.com/OpenC3/cosmos/blob/main/openc3/python/openc3/conversions/packet_time_formatted_conversion.py) conversion 
2. Load the plugin 
 
**Expected behavior**
The system knows how to find and load all built in and provided conversions

**Screenshots**
N/A

**Environment (please complete the following information):**

- OS: ALL 
- OpenC3 COSMOS Version: 6.0.1 & 6.0.2
- Browser: ALL

jmthomas, 2025-04-07:
We have 2 conversions: `packet_time_formatted_conversion` and `packet_time_seconds_conversion`. Please try one of these.

Is there documentation we need to fix to clarify the conversions available?

badrobit, 2025-04-07:
Sorry I think it was confusion due to the ticket naming this is what I have in my `tlm.txt`

```cmake
APPEND_ITEM PACKET_TIME 0 DERIVED "Packet time based on provided timestamp field"  LITTLE_ENDIAN
    READ_CONVERSION openc3/conversions/packet_time_formatted_conversion.py timestamp
```

jmthomas, 2025-04-07:
This worked fine for me:
```
  APPEND_ITEM MY_PACKET_TIME 0 DERIVED "Packet time based on provided timestamp field"  LITTLE_ENDIAN
    READ_CONVERSION openc3/conversions/packet_time_formatted_conversion.py
```
Note that the item `PACKET_TIME` is reserved so do not use that name. Also note that `packet_time_formatted_conversion.py` does not take any parameters.

badrobit, 2025-04-07:
What if we are trying to set the internal `PACKET_TIME` to a timestamp from the payload, e.g. we get an upload of older telemetry and we want all the values to have the time from the packet payload instead of when the payload is parsed? This is why we had been using the reserved `PACKET_TIME`

badrobit, 2025-04-08:
Thanks using the unix formatted time solved the issue!

closed 2025-04-08 by badrobit