OpenC3/cosmos #2007: Recommended way to share screens across targets of the same "type"?

0lionelzhang0, 2025-04-08:
Example: I have a target VEHICLE with cmd_tlm defined with `COMMAND VEHICLE` and `TELEMETRY VEHICLE` etc.

I install this VEHICLE plugin with variable target name CAR1 and create this screen e.g.  
```
VERTICAL
  LABELVALUE CAR1 STATUS TEMP
END
```

If I then install this VEHICLE plugin with variable target name CAR2, is there a way to reuse the screen from CAR1 without creating another duplicate screen? Or is there a way to create screens that are agnostic to variable target name, that could be used across CAR1, CAR2, CAR3, etc?

Thanks!

ryanmelt, 2025-04-08:
Yep.  Everywhere in your config files where you have "VEHICLE" replace it with ```<%= target_name %>``` 
That will substitute in the correct name when the plugin is installed.

0lionelzhang0, 2025-04-09:
Got it. However, it seems if a user were to make a change to this screen from the web UI it would get saved to the targets_modified directory with the hardcoded target name. Then, in order to bring this back into the VEHICLE plugin files, one would need to do a find/replace for CAR1 -> `<%= target_name %>` ? 

I think it would be sweet if there were a way to create scripts and screens from within the COSMOS web code editors that are target_name agnostic while retaining the auto-completion feature. This way if a user makes edits to a script or screen in the web UI they can be easily used across all targets (e.g. CAR1, CAR2, CAR3) without having to mess with the source files and rebuild the plugin.

ryanmelt, 2025-04-11:
I don't anticipate this changing.   Once a plugin is installed, all the internal variables are rendered at install time, and the '''<%= erb %>''' code is replaced with actual values.  Permanent changes should be made to the plugin itself in the git repository where you keep the plugin.

closed 2025-04-11 by ryanmelt