OpenC3/cosmos #2011: microservice error automatically updates without page refresh in Admin console

321github123, 2025-04-10:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Is your feature request related to a problem? Please describe.**
Our microservice catches unexpected errors and sets the self.error field. When clicking the caution sign icon, the message does not update unless we refresh the page

**Describe the solution you'd like**
The message held by the caution sign icon would automatically update.

**Additional context**

[image]

jmthomas, 2025-04-11:
This microservices page automatically updates every 2 seconds and should pull in the latest data? Are you saying that the dialog itself doesn't update or when you click the triangle and create a new dialog, it is still the old data?

321github123, 2025-04-11:
Clicking the triangle after 2 seconds doesn't produce updated data

closed 2025-04-11 by jmthomas