OpenC3/cosmos #2042: Question: Is there a method for caching part of a packet for use on the next pass

badrobit, 2025-04-25:
I have a protocol that splits up a root packet with many smaller packets embedded in it. This is based on the Python protocol base class. I have an internal variable where I can store the packets, but the `read_data` method is only triggered on new data hitting the interface. I would rather not have to rebuild and rebroadcast the remainder of the packet. What ends up happening is that there are more "internal packets" than input packets, so the internal packets end up never getting processed and are abandoned.

ryanmelt, 2025-04-26:
From our protocol docs:

```
On every call to read, an empty string "" is first passed down to each of the read Protocol's read_data() method before new raw data is attempted to be read using the Interface's read_interface() method. This is a signal to Protocols that have cached up more than one packet worth of data to output those cached packets before any new data is read from the Interface. Typically no data will be cached up and one of the Protocols read_data() methods will return STOP in response to the empty string, indicating that more data is required to generate a packet. Each Protocol's read_data() method can return one of three things: data that will be passed down to any additional Protocols or turned into a Packet, STOP which means more data is required from the Interface for the Protocol to continue, or DISCONNECT which means that something has happened that requires disconnecting the Interface (and by default trying to reconnect). Each Protocol's read_data() method is passed the data that will eventually be turned into a packet and returns a possibly modified set of data. If the data passes through all Protocol's read_data() methods it is then converted into a COSMOS packet using the Interface's convert_data_to_packet() method. This packet is then run in a similar fashion through each Read Protocol's read_packet() method. This method has essentially the same return possibilities: a Packet (instead of data as in read_data()), STOP, or DISCONNECT. If the Packet makes it through all read_packet() methods then the Interface packet read counter is incremented and the Packet is returned to the Interface.
```

ie. If read_data receives an empty string, this is your signal to release queued data.

closed 2025-04-26 by ryanmelt

badrobit, 2025-04-28:
That works but only if the source is still connected, if the connection has disconnected but there is still internally queued data or no data incoming then the `read_data` method is not called and the queued data ends up not being worked with. 

How should I trigger or continue working through that internal queue if there is no external driver currently connected but data from previous passes that we need to work through?

badrobit, 2025-04-28:
The connection to the satellite is done, so we are done getting new packets from the satellite, but we still have internal sub-packets that had been pulled out of previous inputs. I know I could send new packets to the socket with the internal packets again, but that seems like a fair amount of overhead, as I need to re-wrap them before sending. This could include a few headers with CRC and crypto checks, so being able to loop through existing data would be helpful. 

The main issue is that I am forced to pass data via a return in the `read_data` method, so I cannot have something sit there and make multiple returns.

ryanmelt, 2025-04-28:
read_data will be repeatedly called with an empty string, as long as you return something when it is called with an empty string.
This allows you to empty an internal queue even if data isn't flowing.

badrobit, 2025-04-28:
I do not see this behavior on COSMOS 6.0.1. Once the connecting system disconnects, no more attempts to call `read_data` are made. (From your description, I would expect the method to continue to be called and an empty byte passed through.) At the end of my protocol stack, it returns just the bytes meant to be parsed by the `tlm.txt` definitions, which does not appear to allow me to return an empty string.

Is there an example you can provide for a protocol where you can both pass the byte data you did find in the packet to the definitions in `tlm.txt` and the empty string to allow for the `read_data` to continue to be called? 

When I add a print statement for whatever is in the incoming `data` on the `read_data` method, no more messages are printed out after the system disconnects, so there are no more opportunities for me to go through my internal data. 

I also have a log for how many internal messages are left. At the time of disconnect, I see more than 50 internal packets have been identified and are awaiting processing. I cannot trigger the `read_data` method to occur again because I have to return the raw packet bytes so the tlm.txt can parse them.

ryanmelt, 2025-04-28:
You don't return an empty string.  read_data is called with data = "" by the interface.  
See: https://github.com/OpenC3/cosmos/blob/main/openc3/lib/openc3/interfaces/interface.rb#L269

Pay attention to the "first" variable, and how the protocols are called.

badrobit, 2025-04-28:
That is where I am confused, as the interface (a TCP connection) has been closed. This wouldn't be the first use of the interface; it would be the last use. 

Our use case in this example is TCP connection (interface) -> protocol 1 (strips out first layer of headers) -> second protocol (finds the subpackets and for now is the "protocol" where the internal list is stored). 

```cmake
INTERFACE <%= target_name %>_INT openc3/interfaces/tcpip_client_interface.py <%= host %> <%= tc_port %> <%= tm_port %> 10.0 None
  MAP_TARGET <%= target_name %>
  PROTOCOL READ protocols/first_protocol.py
  PROTOCOL READ protocols/second_protocol.py
  LOG_STREAM
```

So, while the interface gets TCP packets, `read_data` is called, but once the TCP connection is broken, it does not continue to be called. Does this require me to make a custom interface definition, as it sounds like you are saying anything that works based on [the standard interface](https://github.com/OpenC3/cosmos/blob/main/openc3/python/openc3/interfaces/interface.py#L169). Should continue to emit empty byte data even when disconnected, which is not my experience.

It seems like I might need a custom tcpip that continues to send the `b""` until my internal queues are empty but I am not sure how the interface would know that.

ryanmelt, 2025-04-28:
A disconnected interface will not call read anymore.  So in that case it won't work.   You will need a custom interface that does not mark itself as disconnected (or return nil from the read() method) until you are done for that use case I think.