OpenC3/cosmos #2043: Multiple calls to Python get_packets throws exception

rherbert42, 2025-04-27:
**Describe the bug**
When calling get_packets multiple times, the following excepction occurs:

Traceback (most recent call last):
  File "/project/herbertr/openc3/cosmos/openc3/python/testreadtlm.py", line 18, in <module>
    id_, tlm_data = tlm.get_packets(id_)
  File "/project/herbertr/openc3/cosmos/openc3/python/openc3/api/tlm_api.py", line 441, in get_packets
    for topic, _, msg_hash, _ in Topic.read_topics(lookup.keys(), list(lookup.values()), None, count):
  File "/project/herbertr/openc3/cosmos/openc3/python/openc3/utilities/store_implementation.py", line 201, in read_topics
    result = redis.xread(streams, block=timeout_ms, count=count)
  File "/nas01/users/herbertr/venv/lib64/python3.10/site-packages/redis/commands/core.py", line 3977, in xread
    return self.execute_command("XREAD", *pieces, keys=keys)
  File "/nas01/users/herbertr/venv/lib64/python3.10/site-packages/redis/client.py", line 559, in execute_command
    return self._execute_command(*args, **options)
  File "/nas01/users/herbertr/venv/lib64/python3.10/site-packages/redis/client.py", line 567, in _execute_command
    return conn.retry.call_with_retry(
  File "/nas01/users/herbertr/venv/lib64/python3.10/site-packages/redis/retry.py", line 62, in call_with_retry
    return do()
  File "/nas01/users/herbertr/venv/lib64/python3.10/site-packages/redis/client.py", line 568, in <lambda>
    lambda: self._send_command_parse_response(
  File "/nas01/users/herbertr/venv/lib64/python3.10/site-packages/redis/client.py", line 542, in _send_command_parse_response
    return self.parse_response(conn, command_name, **options)
  File "/nas01/users/herbertr/venv/lib64/python3.10/site-packages/redis/client.py", line 584, in parse_response
    response = connection.read_response()
  File "/nas01/users/herbertr/venv/lib64/python3.10/site-packages/redis/connection.py", line 616, in read_response
    raise response
redis.exceptions.ResponseError: Invalid stream ID specified as stream command argument

**To Reproduce**
Run the following program with the COSMOS demo:

`import time

import openc3.api.tlm_api as tlm
import openc3.api.target_api as target

default_target = 'INST'
pkt_names = tlm.get_all_tlm_names(default_target)
subscribe_list = []
for pkt_name in pkt_names:
    subscribe_list.append([default_target, pkt_name])
    break

id_ = tlm.subscribe_packets(subscribe_list)

while True:
    print(id_)
    id_, tlm_data = tlm.get_packets(id_)
    time.sleep(1)`

Application output before exception:
`(venv) [herbertr@hostname python]$ python testreadtlm.py
DEFAULT__DECOM__{INST}__MECH____1745761759877-0
DEFAULT__DECOM__{INST}__MECH____1745761759877-0
DEFAULT__DECOM__{INST}__MECH____DEFAULT__DECOM__{INST}__MECH____1745761759877-0
`

**Expected behavior**
Telemetry should continue to be retrieved.

The following change fixes the behavior:

`diff --git a/openc3/python/openc3/api/tlm_api.py b/openc3/python/openc3/api/tlm_api.py
index 18283d616..d7eaf7464 100644
--- a/openc3/python/openc3/api/tlm_api.py
+++ b/openc3/python/openc3/api/tlm_api.py
@@ -438,7 +438,7 @@ def get_packets(id, count=1000, scope=OPENC3_SCOPE):
     # Convert it back into a dict to create a lookup
     lookup = dict(zip(items[::2], items[1::2]))
     packets = []
-    for topic, _, msg_hash, _ in Topic.read_topics(lookup.keys(), list(lookup.values()), None, count):
+    for topic, id, msg_hash, _ in Topic.read_topics(lookup.keys(), list(lookup.values()), None, count):
         # # Return the original ID and empty array if we didn't get anything
         # for topic, data in xread:
         # for id, msg_hash in data:`

**Screenshots**
N/A

**Environment:**

- OS: RHEL8
- OpenC3 COSMOS Version 6.3
- Browser: N/A
- Python version: 3.10

ryanmelt, 2025-04-28:
"api" code should not be used outside of internal code like custom microservices.  You need to use the "script" methods.

```
import time
from openc3.script import *

default_target = 'INST'
pkt_names = get_all_tlm_names(default_target)
subscribe_list = []
for pkt_name in pkt_names:
    subscribe_list.append([default_target, pkt_name])
    break

id_ = subscribe_packets(subscribe_list)

while True:
    print(id_)
    id_, tlm_data = get_packets(id_)
    time.sleep(1)
```

jmthomas, 2025-04-30:
@rherbert42 does that make sense? Let me know if the documentation is not clear on this point. Documentation for subscriptions is found here: https://docs.openc3.com/docs/guides/scripting-api#subscribe_packets

ryanmelt, 2025-05-01:
There is a bug in the api version that I fixed in my referenced PR.  The script version works fine though.

closed 2025-05-03 by ryanmelt

rherbert42, 2025-05-07:
Yeah we're basically using this as a microservice to interact with external systems.

PR #2035 (2025-05-03): Goto, Script Status Improvements, Restore Running Script
closes #555
closes #792 
closes #1465
closes #1537
closes #1834
closes #1934 
closes #2043  

Still need to retest python and ruby script control apis, and work through CI/CD failures