OpenC3/cosmos #2046: Is there a way to set a global parameter that is accessible in scripts?

mkalin2, 2025-04-29:
I have COSMOS deployed on three separate systems. I have the same plugin that I am using on all three systems, and would like a way for scripts to be able to check which system they are running on based on a global parameter.

Can I do this using a `VARIABLE` in plugin.txt? Can scripts access variables in the .env file? Something else?

ryanmelt, 2025-04-29:
Most variables in the .env file are available to scripts.   Though that might get more restrictive at some point in the future.   The official way to do this would be to set a global environment variable from the ScriptRunner user interface.  
Script -> Global Environment

mkalin2, 2025-04-29:
How would I get a value from the .env file? With `stash_get` ?

ryanmelt, 2025-04-29:
They should just be environment variables with either option.

In ruby:
```
ENV['VARIABLE_NAME']
```

In python:
```
import os
os.environ['VARIABLE_NAME']
```

mkalin2, 2025-04-29:
Oh that makes sense thanks

closed 2025-04-29 by ryanmelt