OpenC3/cosmos #2058: Is there a way to version control new "Tools" links?

0lionelzhang0, 2025-05-07:
[image]

When adding links with "Admin Console" -> "Tools", is there a way to commit these changes so that it could be deployed across multiple COSMOS instances? Similar to how when changing some settings, they will generate a .json in DEFAULT/settings.

In our case, many of these links point to paths on the same host e.g.  
`https://<cosmos-server>/some-other tool` (

ryanmelt, 2025-05-07:
Plugins can define tools, so creating a plugin with your list of custom tools would be the recommended way to do this.
https://docs.openc3.com/docs/configuration/plugins#tool-1

closed 2025-05-07 by ryanmelt

0lionelzhang0, 2025-05-07:
Got it, thanks!