OpenC3/cosmos #2064: `start` script function does not support absolute filepaths

mbaerwaldt, 2025-05-09:
I think this is a bug, but it also might be a point of clarification in the [documentation of the `start` script function,](https://docs.openc3.com/docs/guides/scripting-api#start) which describes the procedure filename parameter as 

> Name of the test procedure file. These files are normally in the procedures folder but may be anywhere in the Ruby search path. **Additionally, absolute paths are supported.**

**Describe the bug**
When I try to start a script using `start` with an absolute path, I get this error:
```
XMinioInvalidObjectName : Object name contains unsupported characters.
```

**To Reproduce**
My test looks like this:
```ruby
class DC_Power < Cosmos::Test
  def setup
    if defined?($uut_control) == nil
      log_info("Starting initialization because UUT is not initialized!")
      TestUtils.init_test_setup()
    end
  end

  def test_example_dc_power
    report_status("Running #{Cosmos::Test.current_test_suite}:#{Cosmos::Test.current_test}:#{Cosmos::Test.current_test_case}")
    start("/plugins/openc3-cosmos-scripts/targets/SCRIPTS/procedures/tests/example/example_script.rb")
  end
```
The folder structure for the plugin looks like this:
```
> plugins
>> openc3-cosmos-scripts
>>> targets
>>>> SCRIPTS
>>>>> procedures
>>>>>> Test_Suites
>>>>>>> stu_test_suite.rb # the file where the test is located
>>>>>> tests
>>>>>>> example
>>>>>>>> example_script.rb # the script file I want to start
```

Steps to reproduce the behavior:

1. Open the 'stu_test_suite.rb' script in Script Runner.
2. After selecting the suite and group, select the `example_dc_power` script.
3. Click 'Start' button to the right of the script selection
4. See error on this line:
[image]
```
2025/05/09 17:19:25.396 (stu_test_suite.rb:150): XMinioInvalidObjectName : Object name contains unsupported characters.
```
It seems like this error occurs when a leading slash `/` is used.

**Expected behavior**
It runs the script located at the absolute path.

**Additional Info**
I confirmed that the absolute path is correct by adding the following line of code to my `example_dc_power` script:
```
puts File.read("/plugins/openc3-cosmos-scripts/targets/SCRIPTS/procedures/tests/example/example_script.rb")
```
It was able to read the file and print it in the Script Runner console with no issues.

I have also played around with adding the path to the file to my ruby search path, and so far have had no luck. If this turns out to not be a bug, it is probably worth updating the documentation to explain how to correctly start scripts using the absolute path.

**Other Things I Have Tried**
This might be a separate issue, but `start` also does not seem to want to run files in the procedures folder. If I use the following path:
```ruby
start("tests/example/example_script.rb")
```
I get a different error:
```
2025/05/09 17:47:28.883 (stu_test_suite.rb:152): RuntimeError : Unable to retrieve: tests/example/example_script.rb
```
I get that same error if I move `example_script.rb` to the top level of the plugin's `procedures` folder and just run
```ruby
start("example_script.rb")
```
I have tried a lot of things and nothing seems to be working, so I wonder if I am on the wrong track. Any guidance on how to use `start` in COSMOS 6 would be appreciated!

**Environment (please complete the following information):**

- OS: Windows 10
- OpenC3 COSMOS Version 6.3.0
- Browser: Chrome

jmthomas, 2025-05-10:
Can you try `start("SCRIPTS/procedures/tests/example/example_script.rb")`

mbaerwaldt, 2025-05-12:
That worked! Thank you very much!