OpenC3/cosmos #2065: Protocol not found for python plugin

apogeecmb, 2025-05-09:
I'm trying to create a simple test interface to communicate with some test instrumentation using SCPI. Reading through the OpenC3 documention, I found some good examples to get me going down the right path, but I can only get it functioning correctly with a Ruby plugin, not a Python plugin.

Here is a working plugin.txt for a Ruby plugin. Same plugin defined similarly below for python but doesn't work. 
```
VARIABLE test_target_name TEST

TARGET TEST <%= test_target_name %>
INTERFACE <%= test_target_name %>_INT openc3/interfaces/tcpip_client_interface.py 10.0.2.15 5025 5025 10.0 None BURST
  PROTOCOL READ_WRITE TerminatedProtocol 0x0A 0x0A true
  PROTOCOL READ_WRITE CmdResponseProtocol 5.0 0.2 true
```
Is there some other syntax difference in the plugin.txt file between a Ruby and Python plugin that I'm missing?

**Describe the bug**
After creating a new plugin and target, building the plugin, and installing it in OpenC3, I'm seeing the following error in the CmdTlmServer when trying to use the TerminatedProtocol. I see a similar error for other protocols I tried like Length, or if I comment out the Terminated protocol and try it just with the CmdResponseProtocol.

```
Microservice DEFAULT__INTERFACE__TEST_INT dying from exception
Traceback (most recent call last):
  File "/openc3/venv/lib/python3.11/site-packages/openc3/microservices/microservice.py", line 48, in class_run
    microservice = cls(name)
                   ^^^^^^^^^
  File "/openc3/python/openc3/microservices/interface_microservice.py", line 483, in __init__
    self.interface = InterfaceModel.get_model(name=interface_name, scope=self.scope).build()
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/openc3/venv/lib/python3.11/site-packages/openc3/models/interface_model.py", line 174, in build
    klass = get_class_from_module(
            ^^^^^^^^^^^^^^^^^^^^^^
  File "/openc3/venv/lib/python3.11/site-packages/openc3/top_level.py", line 156, in get_class_from_module
    importlib.import_module(f"{parts[0]}", None)
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1140, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'TerminatedProtocol'
```

**To Reproduce**
Steps to reproduce the behavior:

1. Create a new plugin and target.
2. Define plugin.txt like so:
```
VARIABLE test_target_name TEST

TARGET TEST <%= test_target_name %>
INTERFACE <%= test_target_name %>_INT openc3/interfaces/tcpip_client_interface.py 10.0.2.15 5025 5025 10.0 None BURST
  PROTOCOL READ_WRITE TerminatedProtocol 0x0A 0x0A true
  PROTOCOL READ_WRITE CmdResponseProtocol 5.0 0.2 true
```

3. Install plugin.
4. See error in CmdTlmServer.

**Expected behavior**
Functioning plugin with desired interface and protocols.

**Screenshots**

[image]

**Environment (please complete the following information):**

- OS: Ubuntu 22.04 (VirtualBox VM)
- OpenC3 COSMOS Version 6.1.0 - Also saw similar issue on another installation I tried earlier today. I believe it was 5.22.
- Browser: Firefox

ryanmelt, 2025-05-09:
Python requires full paths like you did for the interface.  

```
VARIABLE test_target_name TEST

TARGET TEST <%= test_target_name %>
INTERFACE <%= test_target_name %>_INT openc3/interfaces/tcpip_client_interface.py 10.0.2.15 5025 5025 10.0 None BURST
  PROTOCOL READ_WRITE openc3/interfaces/protocols/terminated_protocol.py 0x0A 0x0A true
  PROTOCOL READ_WRITE openc3/interfaces/protocols/cmd_response_protocol.py 5.0 0.2 true
```

apogeecmb, 2025-05-09:
Thank you for the fix, and thank you for the quick response! I figured it would be user error. Maybe there is a way to make the full path requirement a little more explicit in the documentation, or maybe it's already clear and I just missed it. Thanks again.

closed 2025-05-09 by apogeecmb