OpenC3/cosmos #2129: Issue with Serial Interface on Ubuntu

TeddyPhoenixBoring, 2025-06-10:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Describe the bug**
Sending command to target with serial interface generates an error

**To Reproduce**
Steps to reproduce the behavior:

1. Install plugin with a serial interface defined
2. Send a command from Command Sender
4. See error pop up

**Expected behavior**
Serial command successfully sent.

This previously worked with a slightly modified plugin running on a Windows 11 version of COSMOS using the TCP interface and the standard serial bridge.

**Screenshots**

[image]

[image]

**Environment (please complete the following information):**

- OS: Ubuntu 25.04
- OpenC3 COSMOS Version [e.g. 5.0.6] 6.4.2
- Browser: [e.g. Chrome, Firefox, Safari, Edge] Chrome (Accessed from Windows 11)

ryanmelt, 2025-06-10:
I'd have to see you plugin.txt, but you appear to be mixing Ruby and Python stuff.  Serial interface is only currently supported for Ruby targets.  (Python is coming soon).

TeddyPhoenixBoring, 2025-06-10:
I re-generated a new plugin as a ruby plugin, and copied my plugin.txt and target over and I'm still getting the same error. 
I did delete all python files and changed the language in my target.txt to ruby.

Do I have to do anything else to change the plugin to ruby?

Here is my plugin.txt 
```
# Set VARIABLEs here to allow variation in your plugin

VARIABLE alicat_target_name ALICAT
VARIABLE alicat_host_name host.docker.internal
VARIABLE alicat_port 2951
VARIABLE gas_number 196
VARIABLE alicat_serial /dev/ttyS2

TARGET ALICAT <%= alicat_target_name %>
#INTERFACE <%= alicat_target_name %>_INT openc3/interfaces/tcpip_client_interface.py <%= alicat_host_name %> <%= alicat_port %> <%= alicat_port %> 10.0 None TERMINATED 0x0D 0x0D
#  MAP_TARGET <%= alicat_target_name %>
INTERFACE <%= alicat_target_name %>_INT serial_interface.rb <%= alicat_serial%> <%= alicat_serial%> 19200 NONE 1 10.0 nil TERMINATED 0x0D 0x0D
  MAP_TARGET <%= alicat_target_name %>`
```

closed 2025-06-26 by jmthomas