OpenC3/cosmos #2168: Rootless Podman volume issue

creynol-dev, 2025-06-26:
**Describe the bug**
./openc3.sh run results in an error 

`Error response from daemon: runc: runc create failed: unable to start container process: error during container init: error mounting "/home/cosmos/.local/share/containers/storage/volumes/cosmos_openc3-redis-v/_data" to rootfs at "/data": mount /home/cosmos/.local/share/containers/storage/volumes/cosmos_openc3-redis-v/_data:/data (via /proc/self/fd/6), flags: 0x5026: operation not permitted: OCI permission denied
Error: executing /usr/local/bin/docker-compose -f ./compose.yaml up -d: exit status 1`

Not sure if this is a bug per se or more so just a difference between how rootless podman and docker volumes work. I was able to resolve the issue by converting the volumes for the services in the compose.yml to instead be bind mounts with :z instead of using named docker volumes. Seeing as cosmos is targeting docker it may just make more sense to add a note in the podman setup documentation. https://docs.openc3.com/docs/getting-started/podman 
 

**To Reproduce**
Steps to reproduce the behavior:

1. Create cosmos user for rootless podman
2. Git clone https://github.com/OpenC3/cosmos.git
3. ./openc3.sh run
4. See error

**Expected behavior**
Cosmos containers start

**Environment (please complete the following information):**

- OS: RHEL 8.10
- OpenC3 COSMOS Version 6.5

ryanmelt, 2025-06-26:
Generally the named volumes work with rootless podman.   Permissions are always the hardest part of all this though.   Are you running podman as the "cosmos" user that owns that folder?

creynol-dev, 2025-06-26:
That's correct, I do recall setting this up before as a rootless podman user and it worked just fine but after attempting it again I ran into this issue so I'm unsure as to what changed between then and now. Both occasions were on brand new machines.

jmthomas, 2025-07-25:
Is this a documentation issue and if so where?

Did you solve the permissions problems?

creynol-dev, 2025-07-28:
So I believe on the Non-enterprise version I was able to resolve the issue by switching to bind mounts. However, on the enterprise versions I run into issues with those as well. The postgres container runs into an permission issue as its not able to create it's pgdata folder which in turn means keycloak isn't able to start. 

I was able to eventually get around the OCI permission issue by adding an option to make the volume shared in the compose.yml but then I ran into a more generic permission denied error. I'm assuming this second error is probably related to user namespaces where the containers are running into generic (sub)uid problems now. I wasn't sure how to get around this with docker since I'm used to using podmans --userns=

jmthomas, 2025-12-31:
It sounds like you had luck with a workaround by converting named volumes to bind mounts with the `:z` flag. 

First create local directories:
```bash
mkdir -p openc3-redis-data openc3-redis-ephemeral-data openc3-bucket-data openc3-gems-data
```

Then in compose.yaml, change named volumes like:
```yaml
volumes:
  - "openc3-redis-v:/data"
```
to bind mounts:
```yaml
volumes:
  - "./openc3-redis-data:/data:z"
```

I'm documenting here for posterity. Let me know if this was the solution or if additional documentation / debugging is needed.

closed 2025-12-31 by jmthomas