OpenC3/cosmos #2169: Old Telemetry Definitions being kept

bradenvirdellgenevatech, 2025-06-26:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Describe the bug**
Cosmos appears to be holding onto old TELEMETRY definitions after uploading new gems

**To Reproduce**
Install plugin with serial interface 
test commands 
Update the Plugins Telemetry 
Reupload or upgrade plugin
test more commands
the telemetry with the same ITEM_ID will maintain the old definition including the packet name

**Expected behavior**
The new Telemetry definition should be used not the old telemetry definition

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**

- OS: Ubuntu 24.04
- OpenC3 COSMOS Version 6.4.2
- Browser:  Chrome

Note: This does not happen every single time it seems fairly inconsistent. I have not see this behavior before. The only fix I have found is to shut down the cosmos server and then delete the volumes and restart the server. So it seems to be something wrong in the CMD TLM microservice or potentially something in redis? If you have any suggestions on what to capture if this happens again to help debug I would be more than happy to grab it. Thank you.

bradenvirdellgenevatech, 2025-06-27:
I did some testing to start this morning and got the same issue on my first try
1. Had my current gem up and tested a command and got TLM back
2. Changed just the name of the TLM packet and reuploaded the gem, using the upgrade button in this case
3. Tested the exact same command and expected it to go back to the updated TLM
4. Instead I received the error I was getting before

I then shut down the server, deleted volumes and restarted the server
I uploaded the second gem from the with the changed name and it now works perfectly fine

Looked at the /DEFAULT/target_archives/<target_folder> during both tests and I see the zip files that would hold said archives
I also looked at the /DEFAULT/targets/<target_folder>/cmd_tlm/tlm.txt file and that has the correct packet definition (the updated one).

ryanmelt, 2025-06-28:
Can you provide the error you are receiving?   Also please download and confirm the contents of the _current.zip file are changed.

bradenvirdellgenevatech, 2025-06-30:
[image]

Opened up the zip file and it is showing the correct TLM, also checked the targets folder and it looks correct.

bradenvirdellgenevatech, 2025-06-30:
Nope, 1 plugin on entire server, same plugin with updated version number

jmthomas, 2025-07-09:
@bradenvirdellgenevatech are you reliably able to reproduce? I think we could narrow down the issue if you can reproduce. Please try with the following option after your interface definition:

```
INTERFACE INT_NAME < your params here>
  OPTION SYNC_PACKET_COUNT_DELAY_SECONDS 0 
```

bradenvirdellgenevatech, 2025-07-10:
That seems to have fixed the issue, when I have "SYNC_PACKET_DELAY_SECONDS 0" I no longer have the incorrect packet issue. I will send over operator logs shortly

ryanmelt, 2025-07-16:
This was fixed in COSMOS 6.5.0 by https://github.com/OpenC3/cosmos/pull/2139
which clears out the telemetry counts on undeploy.   
In 6.4.2 old values were left over which causes the issue you are seeing.

closed 2025-07-16 by ryanmelt