OpenC3/cosmos #2198: CLI is clunky

ryan-pratt, 2025-07-10:
We do a log of hard indexing on `ARGV` in the cli, which means that some options arbitrarily require other ones to be set, too. It also doesn't validate extra commands, meaning that if you pass for example `--force`, the cli will happily march along and silently NOT force things because you didn't pass `force` (without the `--`).

Overall, the whole thing is clunky and needs to be reorganized, and start using a proper arg/option/flag parsing utility instead of hard indexing and string comparisons. This probably means breaking changes to our CLI, so target COSMOS v7.0

jmthomas, 2025-07-10:
There are places where we use the Ruby [OptionParser](https://github.com/OpenC3/cosmos/blob/34e3f4bdd60b817b995f55c2a1296dc35dd25f2a/openc3/bin/openc3cli#L754) but there is definitely room for improvement.

ryan-pratt, 2025-10-06:
See #2400 for one of the limitations we need to address

closed 2025-11-03 by mcosgriff