OpenC3/cosmos #2201: Accessing SECRETs in Interface

thomasstjeangenevatech, 2025-07-11:
Hello,

I have a custom interface that requires admin secrets. I have them setup in the plugin that the Interface is declared in, but I couldn't find any documentation on how to actually access them from the interface. Any pointers there would be helpful. Thanks.

jmthomas, 2025-07-11:
The [SECRET](https://docs.openc3.com/docs/configuration/plugins#secret) definitely requires a bunch of pieces. If you define the following:

```
SECRET ENV PASSWORD ENV_PASSWORD OPT_PASSWORD
```

You set the `PASSWORD` secret from the [Secrets](https://docs.openc3.com/docs/tools/admin#secrets) tab in the Admin console.

To use this in your interface you'll then want to implement the `set_option` method and handle the `OPT_PASSWORD` value:

```
    def set_option(option_name, option_values)
      super(option_name, option_values)
      case option_name.upcase
      when 'OPT_PASSWORD'
        @password = option_values[0]
      end
    end
``` 

I think our best example of all of this working together is the [openc3-cosmos-mqtt-test](https://github.com/OpenC3/cosmos/tree/main/examples/openc3-cosmos-mqtt-test) plugin which obviously works with the [MQTT Interface](https://docs.openc3.com/docs/configuration/interfaces#mqtt-interface).

thomasstjeangenevatech, 2025-07-16:
That makes sense, thank you for the info. Are SECRETs accessible from protocols in the same manner?

jmthomas, 2025-07-16:
> That makes sense, thank you for the info. Are SECRETs accessible from protocols in the same manner?

Protocols have access to their interface through the `interface` instance variable.

closed 2025-07-16 by jmthomas