OpenC3/cosmos #2218: How to make Python default language for new scripts?

0lionelzhang0, 2025-07-18:
Is there a way to make .py be the default extension for new scripts in script runner?

ryanmelt, 2025-07-18:
New scripts are not assigned a language, though if the language can't be detected when saved they will default to Ruby.  
The detection looks for import statements, lines that end with :, and f-strings to detect python.

We could potentially add a setting to change to default language, but that doesn't currently exist.

jmthomas, 2025-07-18:
As much as we love our Ruby it might be time to change the heuristics to default to Python unless a Ruby file is detected: `#{`, `require`, `end`, etc. Python is the new hotness. :-)

jmthomas, 2025-09-04:
Create a global option for default scripting language (set to Python). Scripts will still interpret Ruby or Python as possible but anything that can not be distinguished will revert to the default.

This should also change the default syntax for CommandSender disable validation and other places where commands might have keyword parameters.

PR #2354 (2025-09-18): Script default language setting
closes #2218