OpenC3/cosmos #2224: Top banner covers dialog boxes and notifications

0lionelzhang0, 2025-07-20:
[image]

[image]

First screenshot shows the top banner covering the dialog box warning. Also covers notifications in the top right (e.g. Saving...)

COSMOS v6.6.0

clayandgen, 2025-07-24:
Hello! I'm assuming this is the classification banner? I have the classification banner applied:

[image]

And am not able to reproduce the issue on my Macbook using Arc, Chrome, Firefox, and Safari:

[image]

Can you provide some further details on: 
- what banner is used
- machine
- browser

Thanks!

0lionelzhang0, 2025-07-24:
If you make the banner thicker like 40 instead of 20 it'll cover I think.

closed 2025-07-29 by clayandgen