OpenC3/cosmos #2242: CRC protocol should not have "Reflect" dependent on "XOR"

0lionelzhang0, 2025-07-29:
https://github.com/OpenC3/cosmos/blob/main/openc3/python/openc3/interfaces/protocols/crc_protocol.py#L110-L113

There are CRC algorithms that reflect in/out and use XOR of 0x0. As I understand the COSMOS implementation, this is currently not possible.

0lionelzhang0, 2025-07-29:
I think we were passing in Nones as protocol args but didn't realize this would cause args following to not get parsed properly in the order expected.

ryanmelt, 2025-07-30:
Correct, you can't use none/nil for the arguments.

0lionelzhang0, 2025-07-30:
I think this could be made clearer in the documentation. The CRC protocol example has nil/None as arguments.  
https://docs.openc3.com/docs/configuration/protocols

ryanmelt, 2025-07-31:
Its really that you can't mix using defaults with specifying specific values in the current implementation.
All None is fine, all nil is fine, but value nil value isn't fine.

We either update to support mixed in defaults or better document.

PR #2266 (2025-08-07): Update docs for CRC protocol default settings
closes #2242
closed 2025-08-07 by jmthomas