OpenC3/cosmos #2250: Script Runner log in JUnit-compatible format?

jrericks, 2025-07-30:
I looked through the documentation and tickets and don't see an answer to my question. All apologies if i missed it. 

We are using OpenC3 in a gitlab pipeline and collecting the test log output by Script Runner. Since gitlab integrates nicely with junit test reports, we'd like to get the test results from SR into a junit format that can be interpreted by gitlab. I can write a script to convert the SR log to junit format, but figured id check with you guys first to see if something like this already exists. 

Thanks!

jmthomas, 2025-07-30:
Possible dup of #1990 where we're planning on implementing the Common Test Report format. Will that work?

jrericks, 2025-07-31:
Thanks for the quick response. 

I dont think gitlab will be able to read the CTRF formatted report. From what i see online, gitlab only reads junit reports at this time. I'll go ahead and write a script to convert the SR log to junit. 

Question, will the existing SR log format be changing with the CTRF update to Openc3? Or will Openc3 output a new file in the CTRF format, leaving the existing SR log as is? Just want to understand if the CTRF update will break the conversion script that im about to write. 

Thanks!

jmthomas, 2025-07-31:
No we will have an option to enable the CTRF output but will keep our existing output as well.

closed 2025-07-31 by jmthomas