OpenC3/cosmos #2272: Command Sender target dropdown becomes disabled in 6.7.0 if no visible commands for target

0lionelzhang0, 2025-08-07:
[image]

COSMOS v6.7.0

The first target `CFDP_PRIM` only has `HIDDEN` commands so the Target dropdown becomes disabled (new bug introduced in 6.7.0?).

Uninstalling the corresponding plugin allows normal dropdown functionality again.

jmthomas, 2025-08-08:
This makes sense because there are no commands. Or are you saying that the target drop down is disabled when there are other targets that _do_ have commands?

0lionelzhang0, 2025-08-08:
Correct. There is another target with commands. The drop-down being disabled prevents sending any commands from command sender.

PR #2276 (2025-08-17): Do not disable command chooser when command name is null
closes #2272 

@ryanmelt Not sure why you changed this but if you have a target with no commands this effectively disables the command selection permanently.
closed 2025-08-17 by jmthomas