OpenC3/cosmos #2315: Decom Error for Telemetry Packet with BLOCK Item

escheu2, 2025-09-03:
I'm trying to create a telemetry packet that includes several items of type UNIT and one item of type BLOCK.  The item of type BLOCK is truly binary data with no underlying encoding.  The packet appears to be received correctly by OpenC3 and associated with the right telemetry definition, but fails to display within Packet Viewer.  A Decom error is also reported.

```
Decom error: UndefinedConversionError : "\xE2" from ASCII-8BIT to UTF-8
/usr/lib/ruby/gems/3.2.0/gems/json-2.7.5/lib/json/common.rb:314:in `encode'
/usr/lib/ruby/gems/3.2.0/gems/json-2.7.5/lib/json/common.rb:314:in `generate'
/usr/lib/ruby/gems/3.2.0/gems/json-2.7.5/lib/json/common.rb:314:in `generate'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.20.0/lib/openc3/topics/telemetry_decom_topic.rb:45:in `block in write_packet'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.20.0/lib/openc3/utilities/open_telemetry.rb:52:in `in_span'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.20.0/lib/openc3/topics/telemetry_decom_topic.rb:29:in `write_packet'
/openc3/lib/openc3/microservices/decom_microservice.rb:123:in `block in decom_packet'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.20.0/lib/openc3/utilities/open_telemetry.rb:52:in `in_span'
/openc3/lib/openc3/microservices/decom_microservice.rb:87:in `decom_packet'
/openc3/lib/openc3/microservices/decom_microservice.rb:70:in `block (2 levels) in run'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.20.0/lib/openc3/utilities/store_autoload.rb:184:in `block (3 levels) in read_topics'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.20.0/lib/openc3/utilities/store_autoload.rb:182:in `each'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.20.0/lib/openc3/utilities/store_autoload.rb:182:in `block (2 levels) in read_topics'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.20.0/lib/openc3/utilities/store_autoload.rb:181:in `each'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.20.0/lib/openc3/utilities/store_autoload.rb:181:in `block in read_topics'
/usr/lib/ruby/gems/3.2.0/gems/connection_pool-2.4.1/lib/connection_pool.rb:110:in `block (2 levels) in with'
/usr/lib/ruby/gems/3.2.0/gems/connection_pool-2.4.1/lib/connection_pool.rb:109:in `handle_interrupt'
/usr/lib/ruby/gems/3.2.0/gems/connection_pool-2.4.1/lib/connection_pool.rb:109:in `block in with'
/usr/lib/ruby/gems/3.2.0/gems/connection_pool-2.4.1/lib/connection_pool.rb:106:in `handle_interrupt'
/usr/lib/ruby/gems/3.2.0/gems/connection_pool-2.4.1/lib/connection_pool.rb:106:in `with'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.20.0/lib/openc3/utilities/store_autoload.rb:61:in `with'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.20.0/lib/openc3/utilities/store_autoload.rb:177:in `read_topics'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.20.0/lib/openc3/utilities/store_autoload.rb:89:in `public_send'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.20.0/lib/openc3/utilities/store_autoload.rb:89:in `method_missing'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.20.0/lib/openc3/topics/topic.rb:29:in `public_send'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.20.0/lib/openc3/topics/topic.rb:29:in `method_missing'
/openc3/lib/openc3/microservices/decom_microservice.rb:56:in `block in run'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.20.0/lib/openc3/utilities/open_telemetry.rb:52:in `in_span'
/openc3/lib/openc3/microservices/decom_microservice.rb:55:in `run'
/usr/lib/ruby/gems/3.2.0/gems/openc3-5.20.0/lib/openc3/microservices/microservice.rb:55:in `run'
multi_microservice.rb:43:in `block (2 levels) in run'
```

I suspect the issue is related to an invalid UTF-8 character sequence within the BLOCK item but don't understand why that constraint exists.  Is there another way to capture binary data within a telemetry packet?

**Expected behavior**
Packet Viewer should display the BLOCK item as a space delimited hex string regardless of the actual data content.

**Environment:**

- OS: RHEL8
- OpenC3 COSMOS Version 5.20.0
- Browser: Firefox

ryanmelt, 2025-09-03:
What you are showing should work.  We can store any binary data with a BLOCK item.   Things do get serialized to JSON, but our as_json method ensure that raw data gets converted to a compatible format before the JSON formatting.

The code that is crashing is here:
https://github.com/OpenC3/cosmos/blob/v5.20.0/openc3/lib/openc3/topics/telemetry_decom_topic.rb

The as_json with the JSON.generate should prevent the error you are seeing.  

Can you provide an example tlm config that this is happening with?

Also - could you try with 6.7.0 and see if this still occurs?

ryanmelt, 2025-09-03:
Its possible you have installed some other ruby library that is conflicting with our as_json implementation