OpenC3/cosmos #2384: CmdTlmServer packet count sorting doesn't work until packet count was previously displayed

0lionelzhang0, 2025-09-25:
v6.6.0

- Start ingesting different telemetry packet types
- Go to CmdTlmServer -> Tlm packets
- Sort packet count descending
    - Expect to see ingested packets at the top
    - But doesn't actually sort
- Type an expected packet name into search for packet count to show up
- Clear search field
- Try sorting by packet count again, this time the packet count that previously searched for shows up but none of the other expected ones show up

Same behavior observed for Cmd packets sorting

We have on the order of 1000 telemetry packet definitions.

0lionelzhang0, 2026-03-14:
Thank you! Yes, I agree a new tool makes sense.

jmthomas, 2026-08-19:
Remove the sorting and the new tool will implement the functionality requested.