OpenC3/cosmos #2388: Add notification when a script needs user-input

clayandgen, 2025-09-27:
**Description:**
Currently, when a script triggers a popup requiring operator input, there is no clear indication if the operator has navigated to another tab or window. As a result, the operator may be unaware that the script is paused and waiting for input.

**Proposed Solution:**
Introduce a system notification or some other visual indicator (e.g., a badge, flashing icon, or desktop notification) that alerts the operator whenever a script popup occurs. This would ensure operators are immediately aware that their attention is needed, even if they are not actively viewing the script tab.

jmthomas, 2025-10-02:
Might require updates to the notification center to allow more fine grained control of which notifications result in toasts.

ryanmelt, 2025-10-02:
We need to fully implement the "alerts" feature - Notifications that require manual clearing.