OpenC3/cosmos #2391: inject_tlm API method appears to have unstable behavior

henrydix, 2025-09-30:
First check out our main documentation site at [https://openc3.com](https://openc3.com).

**Describe the bug**
When the `inject_tlm` API method is called, a new packet is injected into the interface, but the telemetry item values specified in the call are not updated in the new packet. While debugging this problem in our testing environment, we were able to generate a workaround by calling `override_tlm` first to update the telemetry item values in the CVT first, then call `inject_tlm`.

In our deployment environment where we host Cosmos Enterprise (v6.6.0), we have seen that the `inject_tlm` API method does not perform consistently as documented. When attempting to recreate the bug locally with the open-source 6.6.0 build of Cosmos (using the demo plugin) the problematic behavior was replicated inconsistently, so at minimum, the method is unstable.

**To Reproduce**
Steps to reproduce the behavior:

1. Go to script-runner with openc3 demo plugin installed and run the following a few times with another browser page open with PacketViewer showing the STATUS packet for the EXAMPLE target:
```python
inject_tlm("EXAMPLE", "STATUS", {"VALUE": 1})
inject_tlm("EXAMPLE", "STATUS", {"VALUE": 0})
```
4. You should see that the value for the `VALUE` tlm item does not always update even though a new packet is received (based on the RECEIVED_TIME item value updates).

**Expected behavior**
Expected a new packet to be received by the target interface with the telemetry item values ,specified in the method call, updated appropriately.

**Environment:**

- OS: Kuberneties cluster (deployment)/MacOS Tahoe 26.0.1 (local)
- OpenC3 COSMOS Version: v6.6.0
- Browser: Firefox

henrydix, 2025-10-02:
After further triage and attempts to recreate this bug, I discovered that `override_tlm` had been previously called once on an item in the packet we were trying to inject and our redis container had not been rebooted in our test deployment. This is the root cause of the behavior we were seeing (`normalize_tlm` had not been called and redis topic data had been persisted between tests). Sorry for the confusion on this - I will close this bug-report issue.

jmthomas, 2025-10-02:
You can check script overrides from the Script Runner Script->Overrides menu. We also print out the active overrides in the script log. Glad you found it!